acconeer/acconeer-python-exploration #81: Modify Exemple_sparse_service.c to trigger a MISCGPIO

xmorneau, 2021-05-06:
As the title suggest, I'm having trouble figuring out which variable correspond to the output of the Sparse service. 
For my experimentation, I need to modify the exemple_service_sparce.c in the STM32CubeIDE. I own the XE132 development kit. I want to trigger a MISCGPIO output from the STM32 when the sparce detector pass a certain threshold.
Is there something I'm missing out? How the input can be translated into a boolean output? 
Thanks you for helping me with my issue.

vackraetraed, 2021-05-07:
Hello @xmorneau 

Just to clarify your issue.

1. Are you using "example_service_sparse.c" and want to trigger a GPIO when something happens?
2. Is the problem how to write code to toggle a GPIO on STM32?
3. Is the problem related to writing an algorithm using the sparse data?
4. Does "sparce detector" refer to the sparse service or the presence detector?
5. What are you trying to detect?

Best regards,
Anton from Acconeer :)

xmorneau, 2021-05-07:
Thanks for your quick response.
To clarify my issue:
1. Yes. Or is it the right file to do the modification? I don't want to modify the algorithm from the Sparse service.
2.  I assume that this isn't the right place to find help concerning the programming of an STM32, but I must say I encounter some problems there. I'm not looking for you to answer my concern about the programming, as I will find help in my workplace to find a solution.
3. The algorithm is fine. It fits well for my application. I tested different services from your Exploration Tool program and the «Sparse» service fits my use case.
4. Sorry for bringing confusions from my first comment. I wanted to refer to the «Sparse» service. 
5. The radar must detect quickly that someone is walking underneath it. The sensor will be placed on the ceilling, and must detect, from a range of 20cm to 1m, the presence of a person. The output from the radar must be a signal indicating the presence of a person. In our use case, the sensor must only detect movement underneath it, so I envisage the use a a lens to modify the Half Power Beam Width. From my testing, I tried the  «Sparse» service and the presence detector and It seems to me that the «Sparse» service give a quicker and more accurate result.

Thanks for your support.

vackraetraed, 2021-05-08:
We strongly advice using the presence detector for this since it gives you the binary answer of presence which you are looking for. The presence detector uses the sparse service data to make a decision whether or not there is presence.
You can try and change the filter time constants for faster response.

@erikmansson , got any advice on presence filter parameters for this use case?

Best regards,
Anton

erikmansson, 2021-05-09:
In this case I suggest starting out with only the intra-frame part by setting the `intra_frame_weight` to 1. From there, the `intra_frame_time_const`, `output_time_const` and `detection_threshold` can be adjusted to give the desired behavior. If needed, decrease the time constants to get a faster response. Adjust the detection threshold to make it more or less sensitive.

xmorneau, 2021-05-12:
Thanks for the advices. I don't have any other question. I now have a better comprehension of the service to use and which parameter I should modify to better suit my application. 
 
Sincerely,

Xmorneau

closed 2021-05-12 by xmorneau