acconeer/acconeer-python-exploration #83: Running basic.py on Raspberry Pi

fquitin, 2021-05-14:
I try to run the basic.py script on a Raspberry Pi with an XE132. When I leave everything by default, it works fine. However, when I increase the range_interval in the basic.py script, the code keeps hanging somewhere. When I run it in debug mode, I get the following error: 
| DEBUG | MainProcess | acconeer.exptool.clients.reg.client | recovery attempt failed

When I run the same code on my Ubuntu-laptop, it runs fine without any problem. 

Any idea where this is coming from?

erikmansson, 2021-05-16:
This is likely due to the limited performance of Raspberry Pi:s. As you make the range interval larger, the data rate increases. At some point, the Pi can no longer keep up and you start to experience data loss. The client tries to recover from it, but sometimes fails due to too much loss.

fquitin, 2021-05-17:
Thanks, that was indeed what I was suspecting... 

A couple of follow-up questions: 
- Is there a way to make the interval range larger, but with lower resolution? (i.e. I don't need a data point every 0.4mm) 
- Would running this code with PyPy help in improving the speed? 
- Do you have any experience with single-board computers that are able to keep up with larger data rates? 
Many thanks!

erikmansson, 2021-05-17:
> Would running this code with PyPy help in improving the speed?

Sorry, we haven't tried so we don't know.

> Do you have any experience with single-board computers that are able to keep up with larger data rates?

Not really, we only use the Raspberry Pi.

erikmansson, 2021-05-26:
Closing this for now, feel free to reopen!

closed 2021-05-26 by erikmansson