acconeer/acconeer-python-exploration #84: Run multiple services

kangc0016, 2021-05-26:
Looking at the example_multiple_service_usage.c 

> In a loop:
> Activate the envelope service
> Get the result and print it
> Deactivate the envelope service
> Activate the sparse service
> Get the result and print it
> Deactivate the sparse service
> End of loop

If the envelope service has running average factor set, will the exponential filter being reset when we deactivate the envelope service?

erikmansson, 2021-05-28:
No, the exponential filter is not reset upon activate/deactivate.

We'll get back to you with more information on your other question.

ola1olsson, 2021-05-29:
Hi!

>also how much time does it take to create service, and how much time does it take to activate/deactivate a service?
This is dependent on what MCU you are running on. I think we have log timestamps in most of our example integrations so it should be easy to measure the time by logging before and after. I know it won't be 100% accurate but it won't take exactly the same time always since we are using malloc() and the heap implementation in newlib nano (if you are using this) doesn't allocate/deallocate in constant time.

Don't hesitate to contact us at https://acconeer.freshdesk.com/support/home If you have more questions regarding the software which is not related to exploration tool

closed 2021-06-27 by erikmansson