acconeer/acconeer-python-exploration #85: Fast profile switching for acc_module_server.bin?

dpeterjohnb, 2021-05-27:
I am trying to switch between two configurations for the envelope service on the XM132. I want to grab one full envelope for each configuration and stitch them together.

Running the acc_module_server.bin v2.8 firmware on the XM132, it can take upwards of 178ms to the sensor to be ready to provide envelope data after creating and activating the service through the UART API. 

Using the example_multiple_service_usage.c, I am able to set to set up two different configurations and switch between them under 10ms. 

Is there any functionality in acc_module_server.bin that can get this type of speed in switching configurations or will I have to create my own firmware for the XM132 using the SDK and the example_multiple_service_usage.c as a base?

ola1olsson, 2021-05-29:
>Is there any functionality in acc_module_server.bin that can get this type of speed in switching configurations or will I have to create my own firmware for the XM132 using the SDK and the example_multiple_service_usage.c as a base?

Unfortunately no. Please use the software SDK if you need performance

ola1olsson, 2021-05-29:
Don't hesitate to contact us at https://acconeer.freshdesk.com/support/home If you have more questions regarding the software which is not related to exploration tool

dpeterjohnb, 2021-06-02:
Thanks for your response! I'll close this issue.