acconeer/acconeer-python-exploration #86: Interrupt test failed

yilmaznaslan, 2021-06-10:
Hi,

When you using the Sparkfun module with the most the Rasperry Pi Os Buster, sensor gives the following error. Is there any way to fix this ?
Thank you

> pi@raspberrypi:~/rpi_sparkfun $ ./out/example_assembly_test 
Acconeer software version v2.8.0
00:01:11.236 (I) (rss) Radar system software activated
00:01:11.241 (I) (cpd_communication_test) ASIC identity test passed
00:01:11.262 (I) (cpd_communication_test) Register write-and-read test passed.
00:01:11.286 (I) (cpd_communication_test) Buffer write-and-read test passed
00:01:13.291 (E) (sensor_protocol) Timeout waiting for event on sensor 1
00:01:13.291 (I) (sensor_diagnostics) Run_status stopped 14
00:01:13.292 (I) (sensor_diagnostics) WakeupInput[7..0]_wakeup_config: 0 0 0 0 0 1572 24 0
00:01:13.292 (I) (sensor_diagnostics) Inputmux_wakeupinput[7..0]_config: 0 0 0 0 0 0 13 0
00:01:13.293 (I) (sensor_diagnostics) Timer[3..0]_wakeup_config: 0 0 0 0
00:01:13.293 (I) (sensor_diagnostics) asic_error_status: 0
00:01:13.293 (I) (sensor_diagnostics) Adc_too_fast: 0, Adc_fifo_full: 0, Spi_missing_ack: 0, Spi_unknown_cmd: 0, Stack_underflow: 0, Stack_overflow: 0
00:01:13.294 (I) (sensor_diagnostics) Stack_level_status_register: 0
00:01:13.294 (I) (sensor_diagnostics) Stack level 0: Addr=14
00:01:13.294 (I) (sensor_diagnostics) Stack level 1: Addr=196 Loop=0
00:01:13.294 (I) (sensor_diagnostics) Stack level 2: Addr=195 Loop=0
00:01:13.294 (I) (sensor_diagnostics) Stack level 3: Addr=0 Loop=0
00:01:13.294 (I) (sensor_diagnostics) Stack level 4: Addr=0 Loop=0
00:01:13.294 (I) (sensor_diagnostics) Stack level 5: Addr=0 Loop=0
00:01:13.294 (I) (sensor_diagnostics) Stack level 6: Addr=0 Loop=0
00:01:13.294 (I) (sensor_diagnostics) Stack level 7: Addr=0 Loop=0
00:01:13.294 (I) (sensor_diagnostics) asic_identity_number: 4370, interrupt_status: 88, Scratchpad Config 7: 8193
00:01:13.295 (I) (sensor_diagnostics) Wakeup_status0_register: 12
00:01:13.295 (I) (sensor_diagnostics) vco_cbank_config: 0
00:01:13.295 (I) (sensor_diagnostics) pll_config: 0
00:01:13.295 (I) (sensor_diagnostics) Pll_div_config: 11
00:01:13.295 (E) (cpd_interrupt_test) The first interrupt test failed, please check either HW connection or HAL declarations
00:01:13.295 (E) (cpd_communication_test) Interrupt test failed
00:01:13.295 (I) (cpd_communication_test) Effective bit-rate: 5588 kbps
00:01:13.295 (I) (cpd_communication_test) SPI frame overhead: 62 us
Name: communication_read, result: Pass
Name: communication_write_read, result: Pass
Name: communication_interrupt, result: Fail
Name: supply, result: Fail
Name: clock, result: Fail
Assembly test: There are failed tests
00:01:13.295 (I) (rss) Radar system software deactivated

vackraetraed, 2021-06-10:
Hello @yilmaznaslan ,

If you are using the red Sparkfun, then it should not be used directly with the Raspberry Pi since it may damage the GPIOs permanently on the Raspberry.
Instead you should wire the pins to the Raspberry and make sure that 5V is connected to the 3.3V instead of 5V.
Raspberry needs to use 3.3V logic rather than 5V (this is due to faulty design by Sparkfun)

yilmaznaslan, 2021-06-10:
Thank you for the quick response @vackraetraed ,

However I have been using the red Sparkfun for months already, suddenly is stopped working. Besides I checked the gpio pins on raspberry pi and I coulnd't detect any anormalies on outtput and input signals. Is there anyway to carry out calibration on the sensor or any method to determine the source of the error.

Thank you

vackraetraed, 2021-06-10:
Hello again @yilmaznaslan 

My recommendation is still to wire the signals correctly on the Raspberry, i.e. use 3.3V instead of 5V.
I can not rule out that this is not the issue.
Currently it looks like RSS times out waiting for interrupt from the sensor.
Since the Sparkfun breakout board outputs 5V to the Raspberry (which the Raspberry does not support), it might stop working at any point or stop working with increasing temperature, or any number of things.

yilmaznaslan, 2021-06-11:
I have applied your suggestions but unfortunately the error remains same. I have also tested this solution on various Raspberry Pi's but resulted in the same error message. I believe the source of this error is more likely to be on sensor's  interrupt pin not on raspberry pi

vackraetraed, 2021-06-11:
I see, I assume that in all cases the SPI communication passes just like the first log you sent?
I might have to consult our HW team about this issue and get back to you.

vackraetraed, 2021-06-11:
Meanwhile, do you have the possibility to perhaps hook up the interrupt pin to an oscilloscope or a logic analyzer of some kind and monitor the interrupt pin when running the firmware?

yilmaznaslan, 2021-06-11:
Okay,

thank you for the support. Once I check it with logic analyzer, I will keep here updated

closed 2021-06-22 by yilmaznaslan