acconeer/acconeer-python-exploration #87: Random Timeout Error with multiple sensors

yilmaznaslan, 2021-07-01:
**Describe the bug**
Hello,

When using XC112 with multiple XR112, service create function sometimes fail sometimes succeed. I am using the same config for all of the cases and didn't face any problem when using single sensors. However in the case of two sensors it randomlly fails this following error when creating a iq service

> Acconeer software version v2.8.0
> 13:38:57.528 (I) (rss) Radar system software activated
> acc_rss_activate() has succesfull
> Creating iq service config for sensor 2
> Creating iq service config for sensor 4
> acc_config_radar_services() has succesfully terminated
> 13:38:57.528 (I) (base_configuration) sensor 2 config: 30 11 2 7 1 ACTIVE A 0 0 0
> 13:38:57.553 (I) (cpd_cbank_and_vana_calibration) Result: (4, 0)
> 13:38:57.679 (I) (dll_calibration) Result: (2, 5, 50, 27, 998, 1112, 15, false)
> 13:38:57.679 (I) (radar_engine_staggered) Sensor calibration successful
> 13:38:59.704 (E) (sensor_protocol) Timeout waiting for event on sensor 2
> 13:38:59.704 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.705 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.706 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.707 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.708 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.709 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.709 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.709 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.709 (I) (sensor_diagnostics) Run_status running 633
> 13:38:59.709 (I) (sensor_diagnostics) Run_status running 633
[... 50 lines cut ...]
> 13:38:59.714 (I) (sensor_diagnostics) Adc_too_fast: 0, Adc_fifo_full: 0, Spi_missing_ack: 0, Spi_unknown_cmd: 0, Stack_underflow: 8, Stack_overflow: 0
> 13:38:59.715 (I) (sensor_diagnostics) Stack_level_status_register: 0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 0: Addr=633
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 1: Addr=632 Loop=0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 2: Addr=140 Loop=0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 3: Addr=77 Loop=0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 4: Addr=97 Loop=0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 5: Addr=96 Loop=0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 6: Addr=96 Loop=0
> 13:38:59.715 (I) (sensor_diagnostics) Stack level 7: Addr=0 Loop=0
> 13:38:59.716 (I) (sensor_diagnostics) asic_identity_number: 4370, interrupt_status: 112, Scratchpad Config 7: 2054
> 13:38:59.716 (I) (sensor_diagnostics) Wakeup_status0_register: 6
> 13:38:59.716 (I) (sensor_diagnostics) vco_cbank_config: 4
> 13:38:59.716 (I) (sensor_diagnostics) pll_config: 2171
> 13:38:59.716 (I) (sensor_diagnostics) Pll_div_config: 11
> acc_service_create() for sensor 2 has failed
> 

**Environment (please complete the following information):**
 - Host OS: Rasbian
 - Acconeer hardware:  Raspberry Pi XC112 + XR112
 - Acconeer software version: 2.8.0

Note;
Since I couldn't find a repository for the source files, I posted here even though I didn't use the exploration tool.

vackraetraed, 2021-07-02:
Hello @yilmaznaslan ,

What configurations are you setting for the sensors?
Can you share the code when you setup and run the sensors?
Does it only happen with sensor 2? What happens if you use sensor 1 and 4?

erikmansson, 2021-07-14:
Closing this due to inactivity, feel free to reopen!

closed 2021-07-14 by erikmansson