acconeer/acconeer-python-exploration #89: No response on Serial port on MacOS

talha196, 2021-07-20:
I am trying to connect XM132 and XM112 via UART. In the Exploration GUI, both ports show up but fail to connect. Both of them work fine on Linux though.

Here's a snippet of the error with xm112 with xb112 eval board
Traceback (most recent call last):
  File "gui/main.py", line 1558, in connect_to_server
    info = self.client.connect()
  File "/Users/tz/Library/Python/3.7/lib/python/site-packages/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/Users/tz/Library/Python/3.7/lib/python/site-packages/acconeer/exptool/clients/reg/client.py", line 214, in _connect
    raise ClientError("could not connect, no response")
acconeer.exptool.clients.base.ClientError: could not connect, no response

or the with Traceback (most recent call last):
  File "gui/main.py", line 1558, in connect_to_server
    info = self.client.connect()
  File "/Users/tz/Library/Python/3.7/lib/python/site-packages/acconeer/exptool/clients/base.py", line 30, in connect
    info = self._connect()
  File "/Users/tz/Library/Python/3.7/lib/python/site-packages/acconeer/exptool/clients/reg/client.py", line 223, in _connect
    self._handshake()
  File "/Users/tz/Library/Python/3.7/lib/python/site-packages/acconeer/exptool/clients/reg/client.py", line 425, in _handshake
    self._link.recv_until(exp_frame)
  File "/Users/tz/Library/Python/3.7/lib/python/site-packages/acconeer/exptool/clients/links.py", line 175, in recv_until
    raise LinkError("recv timeout")
acconeer.exptool.clients.links.LinkError: recv timeout 

macOS version is 11.4

vackraetraed, 2021-07-20:
Hello @talha196 

Which version of the exploration tool are you using?

talha196, 2021-07-20:
> Hello @talha196
> 
> Which version of the exploration tool are you using?

Hello @vackraetraed 
I cloned the latest project from the github, so the latest one

vackraetraed, 2021-07-20:
Is the second error when you are using XM132?

talha196, 2021-07-20:
> Is the second error when you are using XM132?

Yes. The first one is the XB112  eval board

vackraetraed, 2021-07-20:
All right, all we can say for now is "make sure you have the proper drivers installed for OSX".
For XM112 see: https://ftdichip.com/drivers/vcp-drivers/
For XM132 see: https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

We do not really support OSX. We have made a few attempts at porting it that have been successful for certain customers to run exploration tool on OSX.

talha196, 2021-07-20:
> https://ftdichip.com/drivers/vcp-drivers/

Thank you for your response. I have installed these drivers and the devices do show up in the list. Unfortunately the error still persists

vackraetraed, 2021-07-20:
You should probably try and verify that the module software is actually running on the module by trying to capture the debug log. Does it give any output when you try to connect?

If you want to try and debug the python3 serial communication on OSX then you can always try and construct a simple python script that opens a serial connection and just sends 1 read command to the module software. For example: 0xCC 0x01 0x00 0xF8 0x10 0xCD , this should ask for the Product ID from the module.

vackraetraed, 2021-07-20:
Never mind the first part to verify if the module software is running. You said it works on Ubuntu so the module software works.

talha196, 2021-07-20:
DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x03 val: 00 00 00 00
15:36:29.776 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | handshake failed at 115200 baud
15:36:29.980 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x03 val: 00 00 00 00
15:36:30.581 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | handshake failed at 1000000 baud
15:36:30.785 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x03 val: 00 00 00 00
15:36:31.386 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | handshake failed at 3000000 baud
15:40:18.320 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x03 val: 00 00 00 00
15:40:18.324 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | handshake succeeded at 115200 baud
15:40:18.324 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg r req: addr: 0x12
15:40:18.328 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | recv reg r res: addr: 0x12 val: 40 42 0f 00
15:40:18.328 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | switching to 1000000 baud...
15:40:18.328 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x07 val: 40 42 0f 00
15:40:18.332 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | recv reg w res: ok
15:40:18.540 | DEBUG   | MainProcess      | acconeer.exptool.clients.reg.client  | sent reg w req: addr: 0x03 val: 00 00 00 00

erikmansson, 2021-08-09:
Hi @talha196, did you manage to solve the issue?

closed 2021-08-19 by erikmansson