acconeer/acconeer-python-exploration #90: Socket Connection Issue

ingleajinkya, 2021-08-02:
Hi,

1. I am getting a issue - "_no sensor connected, check connections"_ when I am trying to connect XR112 sensor with XC112 connection board through Socket. (Acconeer Exploration GUI)
2. This issue comes randomly. The sensors were working good suddenly this issue got raised.
3. I need your help in understanding what could get wrong.

Regards,
Ajinkya Ingle

erikmansson, 2021-08-09:
Hi! Can you send the output from the streaming server running on the Pi from when this happens?

ingleajinkya, 2021-08-10:
[Streaming Server File.txt](https://github.com/acconeer/acconeer-python-exploration/files/6959181/Streaming.Server.File.txt)
Please find the attached text file

erikmansson, 2021-08-10:
It looks like the Pi is not able to connect to the sensor(s) at all here. Is the issue completely random? Does it happen during active measurements?

ingleajinkya, 2021-08-10:
No, it did not happened during active measurements.

ingleajinkya, 2021-08-12:
Yes the issue is random

vackraetraed, 2021-08-12:
@ingleajinkya 

Is it possible for you to provide a picture of your connection? 
Specifically how the flex cable is connected from XR112 and XC112.

Best regards,
Anton

ingleajinkya, 2021-08-13:
Sensor XR112 is connected by a ribbon cable to XC112 board.
As mentioned in the attached picture - first socket got damaged due to
repeated usage of it.

vackraetraed, 2021-08-13:
@ingleajinkya 

I can not see any attached picture. However, the main reason that I am asking is because 99% of the time this issue happens because the flex cable is turned upside-down in one of the sockets.

Best regards,
Anton

ingleajinkya, 2021-08-16:
I also checked with the cables side while plugging it. I did plugged it
properly. But still this issue arises again.
Is it possible that their is some issue with sockets?

vackraetraed, 2021-08-17:
Hello,

It could really be anything that might be broken then.
Do you have another XR112 and flex cable to try, in order to try and narrow down the issue.

Otherwise I'd recommend trying to capture the signals from XC112 to XR112 using a logic analyzer or something and try and see if anything looks strange, because currently it simply looks like the radar is not responding. That might be caused by several things, perhaps ENABLE is not turning the radar on or MOSI is damaged or MISO is damaged. Perhaps the radar is not getting proper supply?

Best regards,
Anton