acconeer/acconeer-python-exploration #91: XC112/XR112

Wayne-xixi, 2021-08-16:
Hello, according to the datasheet of xc112, it must be used with Raspberry Pi. Is there already use case in Linux based system without raspberry? Besides, have you developed a package for ROS System? 

Thank you in advance!
best regards
Wayne

vackraetraed, 2021-08-17:
Hello Wayne,

XC112 does not have to be used with Raspberry. I have previously connected XC112 to a Nucleo-L476. There is an example of it in this document: [STM32CubeIDE guide](https://developer.acconeer.com/download/stm32cubeide-user-guide-pdf/)
Though the integration only use one sensor.

Generally we simply provide pre-compiled libraries with example integrations and example applications. If a customer is working on an MCU we do not have experience with, we let the customer write the SPI communication function themselves.

I believe there exists customers who use Linux systems like OpenWrt.

Wayne-xixi, 2021-08-17:
> Hello Wayne,
>
> XC112 does not have to be used with Raspberry. I have previously connected XC112 to a Nucleo-L476. There is an example of it in this document: [STM32CubeIDE [...]

Hello,

thank you very much for the reply. Have you already some use cases with ROS-Framework? I mean ros-package or ros-driver for the evk.

vackraetraed, 2021-08-17:
Hello Wayne,

We do not have anything like that, but our libraries do not really need to know what operating system is running since all the lower level functions can be implemented by the user for their specific OS.
It is more important for us to know what hardware architecture and processor it is going to run on.

Best regards,
Anton

vackraetraed, 2021-08-17:
For example, the Raspberry PI libraries have been compiled with aarch64, which is 64 bit ARM processors. So if you are using a 64 bit ARM processor then the Raspberry PI libraries should work.

Best regards,
Anton

Wayne-xixi, 2021-08-18:
> For example, the Raspberry PI libraries have been compiled with aarch64, which is 64 bit ARM processors. So if you are using a 64 bit ARM processor then the Raspberry PI libraries should work.
> 
> Best regards,
> Anton

Hi Anton,

thank you very much for the nice reply. It's very exciting to know that the libraries are independent of the os.  But XC112 does really need a 40 pin to connect with Raspberry or Nucleo-L476.  I want to use the XC112 directly with NVIDIA® Jetson™ TX2 so that the sensor data can be used by the robot navigation stack. This means I use USB or I2C instead of 40 Pins. Is there a solution? 

Kind regards,
Wayne

vackraetraed, 2022-01-21:
Hello @Wayne-xixi 

Sorry for the late answer but our radar uses SPI, so in order to use XC112 to communicate with our radar sensors, then SPI is needed. You could try and use one of our modules that can use I2C or UART of USB to communicate with the processor that in turn uses SPI commands to communicate with the sensor.

Wayne-xixi, 2022-01-24:
Hi,
thanks very much, I have fixed it. btw, I have written a ros driver for xc112 and xr112 for robotics application if anyone needs I can also upload it. 
Cheers,
Wayne

closed 2022-01-24 by vackraetraed