acconeer/acconeer-python-exploration #92: XC112/XR112 Sensors

Wayne-xixi, 2021-08-17:
Hello, I used 2 XC112 sensors with XR112. In the Obstacles Detection (IQ) mode, if I show fusion 2 Sensors, the following error came when sensors detect an obstacle.  

`Failed to communicate with server!

main.py
2279
'use_as_noise_filter'` 

The output of terminal:
`Traceback (most recent call last):
  File "gui/main.py", line 2279, in run
    _, record = self.radar.process(sweep, info)
  File "/home/acconeer-python-exploration/gui/data_processing.py", line 97, in process
    out_data = self.external.process(in_data, in_info)
  File "/home/acconeer-python-exploration/examples/processing/obstacle_detection.py", line 472, in process
    fused_obstacles, shadow_obstacles = self.fusion_handle.update(fused_obstacles)
  File "/home/acconeer-python-exploration/examples/processing/obstacle_detection.py", line 1507, in update
    self.update_fused_obstacle_data(new_fused_obstacle_data)
  File "/home/acconeer-python-exploration/examples/processing/obstacle_detection.py", line 1717, in update_fused_obstacle_data
    self.update_fused_obstacle(idx_fused, new_obstacles[idx_new], idx_new)
  File "/home/acconeer-python-exploration/examples/processing/obstacle_detection.py", line 1752, in update_fused_obstacle
    if self.fusion_params["use_as_noise_filter"]:
KeyError: 'use_as_noise_filter'
/home/.local/lib/python3.6/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py:927: RuntimeWarning: All-NaN slice encountered
  self.bounds[ax] = (np.nanmin(d) - self._maxSpotWidth*0.7072, np.nanmax(d) + self._maxSpotWidth*0.7072)
`

Thanks,
Wayne

Wayne-xixi, 2021-08-18:
Screenshot attached:
[image]

elinayrli, 2021-08-18:
Hi Wayne, thanks for your report! The issue has been resolved in the referenced commit

Wayne-xixi, 2021-08-18:
> Hi Wayne, thanks for your report! The issue has been resolved in the referenced commit

Hi, thanks for your info. It works. 

But I have another question, the estimation of angle in obstacle detection is based on robot velocity. But in the GUI, there is only an absolute value. Obviously, this will have an impact on the measurement accuracy of obstacles. Do I need a vector to define different directions and values of velocity in the real robot test?

erikmansson, 2021-08-19:
Hi!

The obstacle detector cannot handle changing direction, so when turning, reset the detector.

The Python detector does not "natively" handle changing the speed, but the attribute `self.robot_velocity` may be changed during a session. The corresponding detector in the C SDK has a function `acc_detector_obstacle_radial_velocity_to_degrees` which takes the current speed.

Wayne-xixi, 2021-08-19:
Hello Erik,

thank you for your reply.

which C SDK version do you mean? I only found the  acconeer_rpi_xc112 v2.8.2 in the Acconeer Developer Website. But there is only an obstacle detector as lib,  and example using obstacle detector. Where can I find the source C file? And another question, in the python-exploration, there is only fusion for 2 sensors. Is there a possibility for 4 sensors? And it seems that, in the rpi_xcc112 package, there are few places where I can modify myself. In my use case, I want to use this sensor set (4- 8 sensors) in robot, and I need the interface with ROS Middleware to publish the sensor data to the navigation stack. 

Thanks,
kind regards,
Wayne

Wayne-xixi, 2021-08-19:
> Hello Erik,
>
> thank you for your reply.
>
> which C SDK version do you mean? I only found the acconeer_rpi_xc112 v2.8.2 in the Acconeer Developer Website. But there is only an obstacle detector as lib, [...]

@erikmansson @elinayrli

erikmansson, 2021-08-23:
Hi,

The C SDK detectors are closed source, but do implement the exact same algorithms as the corresponding detectors here in Exploration Tool.

Currently, the obstacle detector only supports fusion of 2 sensors. Unfortunately, I don't believe there is any simple way of supporting fusion of more sensors at the moment.

Wayne-xixi, 2021-08-23:
> Hi,
>
> The C SDK detectors are closed source, but do implement the exact same algorithms as the corresponding detectors here in Exploration Tool.
>
> Currently, the obstacle detector only supports fusion [...]

Hi,

thank you for the info. Since the C SDK is a closed source, how could I modify the parameter of the sensor? I mean, with the python-exploration-tool it's very easy to modify the parameter (like `Downsample Scale` `Thresholds`) in the gui.  But in the C SDK, it seems no possibility.

erikmansson, 2021-08-23:
You're right, some parameters cannot be changed in the C SDK. In many cases, it's because we recommend not moving away from their default values. If you need to modify them or have questions about them, please discuss with our customer support team by submitting a ticket [here](https://acconeer.freshdesk.com/support/home).