acconeer/acconeer-python-exploration #93: Count the reps of breathing and respiration pattern

raghavendrajain, 2021-08-22:
I have bought pulsed coherent radar module XM112 and XB112 board. I need some guidance on where to start for my application. 

**Context**
I'm making a breathing app, in which different styles of breathing exercises are to be taught to the users. For example, in one exercise the inhaling has to be done for 3 seconds, holding for 5 seconds and exhalation for 3 seconds. In another exercise, inhalation is to be done with normal speed but exhalation has to be done very fast. There are more than 10 types of breathing patterns. 

For all these exercises, the person is sitting either on chair facing a wall or sitting on the ground in lotus position. 

**Expected solution**
I'd like to 
(a) correct user using some verbal/sound feedback when his breathing pattern varies from the goal pattern
(b) count reps of breathing patterns

I looked into the GUI for Breathing and Sleep Breathing. I think the absolute movement of chest movement would provide me what I need. Please provide your insights into what should I look into. Thanks a lot!

erikmansson, 2021-08-23:
Hi! I think a good place to start would be the `breathing_history` returned from the breathing example processor:

https://github.com/acconeer/acconeer-python-exploration/blob/73d06aa5175a205747a0a84999a94401a223e90a/examples/processing/breathing.py#L228

It contains filtered estimates of the chest movement, seen in the lower right plot.

raghavendrajain, 2021-08-24:
@erikmansson Thank you! Do you mean the plot titled `Breathing Movement` and gives me BPM and depth? I believe this gives me the absolute movement of the distance of the chest from the sensor. Although I expect the user to not move, there may be some small movements while breathing, so how do I filter the movement of anything that is not chest movement? Thank you!

erikmansson, 2021-08-24:
> Do you mean the plot titled `Breathing Movement` and gives me BPM and depth?

Yes

> [...] how do I filter the movement of anything that is not chest movement?

This may be a bit tricky. As a start, you could play around with the filter parameters defined here:

https://github.com/acconeer/acconeer-python-exploration/blob/ce5f4cdf18b820934650050b027b4f677847bc8a/examples/processing/breathing.py#L103

Another approach could be to try to detect if the user is moving too much and temporarily ignore the movement.

raghavendrajain, 2021-08-24:
Thank you very much @erikmansson.

erikmansson, 2021-08-24:
No problem. I'll close this for now but feel free to reopen or create a new issue if you have any more questions.

closed 2021-08-24 by erikmansson