acconeer/acconeer-python-exploration #95: Unexpected interrupt from one of the two A111 radars

Alvedro, 2021-11-04:
In my setup, I have two Sparkfun boards with A111 radars connected via SPI to STM32H7.  The boards are powered from a 3V3 source. There are times when everything works well, but sometimes one of two sensors has a problem with the interrupt pin.

[image]
[image]

You can see a problem at the end of the D6 channel. While uC is communicating with the black sensor, the red sensor pulls down his own interrupt pin. After that, function `acc_detector_distance_get_next` returns only false and interrupts from both sensors are stucked. Sometimes this happens after 3 minutes of communicating, sometimes it can work 20 minutes - there is no rule. Any idea what is going wrong?

vackraetraed, 2021-11-05:
Hello @Alvedro ,

Do you have the possibility to perhaps share your project?
I am mainly interested in the application source file and integration file.
Then I can try and setup a similar project on a Nucleo-STM32H7 and see if I can recreate the issue.

Best regards,
Anton

erikmansson, 2022-01-02:
Closing due to lack of activity, but feel free to reopen

closed 2022-01-02 by erikmansson