acconeer/acconeer-python-exploration #98: Random lines plotted in Acconner Exploration GUI

yahyatawil, 2022-01-09:
**Describe the bug**
The are some random lines plotted in Acconner Exploration GUI when the Obstacle Detection (IQ) is selected 

**To reproduce**
Steps to reproduce the behavior:
0. With Module server firmware flashed to XM112. 
1. Go to Acconner Exploration 
2. Click on Scan Control > Obstacle Detection (IQ)
3. Start Measurment and hover with some obstacles 

**Traceback**
logs from consol:

```
C:\Users\DELL\Desktop\acconner\acconeer-python-exploration-master>python gui/main.py
12:24:26.969 | INFO    | MainProcess      | acconeer.exptool.clients.base        | reported version: 2.10.0
12:24:43.295 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | data length: None
12:24:43.295 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | buffer size: None B
12:25:35.779 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | data length: 21
12:25:35.779 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | buffer size: 42 B
12:25:35.779 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | requested data rate: 0.02 Mbit/s
12:26:03.077 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | data length: 827
12:26:03.078 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | buffer size: 3308 B
12:26:03.078 | INFO    | MainProcess      | acconeer.exptool.clients.reg.client  | requested data rate: 1.75 Mbit/s
```

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Host OS: Windows 10
 - Acconeer hardware: XM112 + XB112
 - Acconeer software version: RSS V2.10.0
 - Exploration version: 2d4f5c0

**Additional context**
N/A

andrehedesand, 2022-01-11:
Hi @yahyatawil!

Thanks for the Issue, we'll look into it and get back to you in this thread.

BR, André

andrehedesand, 2022-01-14:
Hi again @yahyatawil!

I am unable to reproduce this behaviour. Are you able to supply me with more information? I'm interested in:

* Python version (Output from `python --version`)
* Package versions (Output from `python -m pip freeze -r requirements.txt`)
* Windows OS build number (Do the Windows search "About your PC", scroll down to "Windows specifications")

yahyatawil, 2022-01-23:
Please note that the lines appear after few seconds not directly.

The requested information: 

Python 3.9.7

OS Name	Microsoft Windows 10 Pro
Version	10.0.18362 Build 18362

astunparse==1.6.3
cachetools==4.2.4
certifi==2021.10.8
charset-normalizer==2.0.10
click==8.0.3
colorama==0.4.4
cycler==0.11.0
flatbuffers==2.0
fonttools==4.28.5
gast==0.4.0
google-auth==2.3.3
google-auth-oauthlib==0.4.6
google-pasta==0.2.0
grpcio==1.43.0
idna==3.3
importlib-metadata==4.10.0
itsdangerous==2.0.1
Jinja2==3.0.3
joblib==1.1.0
keras==2.7.0
Keras-Preprocessing==1.1.2
kiwisolver==1.3.2
libclang==12.0.0
Markdown==3.3.6
MarkupSafe==2.0.1
oauthlib==3.1.1
opt-einsum==3.3.0
Pillow==9.0.0
protobuf==3.19.1
pyasn1==0.4.8
pyasn1-modules==0.2.8
pyparsing==3.0.6
PyQt5-Qt5==5.15.2
PyQt5-sip==12.9.0
python-dateutil==2.8.2
requests==2.27.1
requests-oauthlib==1.3.0
rsa==4.8
scikit-learn==1.0.2
six==1.16.0
tensorboard==2.7.0
tensorboard-data-server==0.6.1
tensorboard-plugin-wit==1.8.1
tensorflow==2.7.0
tensorflow-estimator==2.7.0
tensorflow-io-gcs-filesystem==0.23.1
termcolor==1.1.0
threadpoolctl==3.0.0
typing_extensions==4.0.1
urllib3==1.26.8
Werkzeug==2.0.2
wrapt==1.13.3
zipp==3.7.0

yahyatawil, 2022-01-23:
btw, `python obstacle_detection.py -u COM3` is not giving similar behaviour

andrehedesand, 2022-01-24:
Hi again @yahyatawil !

Great, thanks for the version dump this will help to troubleshoot the issue!

> btw, `python obstacle_detection.py -u COM3` is not giving similar behaviour

Alright, thanks for this. Great to know!

closed 2022-03-23 by AndersBuhl