acconeer/acconeer-python-exploration #100: UART Connection

kmoretina, 2022-01-12:
I was wondering if it is possible to run a service without using the command window? I am wanting to run iq.py on the XM 112 by being able to press F5 in Python 3.9 rather than using the command window, but I am getting the error below

usage: iq.py [-h] (-u [port] | -s address | -spi) [--sensor id [id ...]]
             [-v | -vv | -q]
iq.py: error: one of the arguments -u/--uart -s/--socket -spi/--spi is required

I understand what the error is, but I was wondering if there is a known way to get around this? I am trying to use a UART connection.

andrehedesand, 2022-01-14:
Hi @kmoretina

Thank you for the question, I'm sure someone will wonder the same in the future!

In your specific case, I would do the following:

1. Make a copy of `iq.py` and move it outside the `examples/` folder
    (either to `acconeer-python-exploration/` folder or wherever you want!)
2. (Optional) Rename it; maybe to `my_iq.py`.
3. Uncomment or remove these lines in your copy (I've commented them out):

```py
...

def main():
    # args = et.utils.ExampleArgumentParser().parse_args()
    # et.utils.config_logging(args)

    # if args.socket_addr:
    #    client = et.SocketClient(args.socket_addr)
    # elif args.spi:
    #    client = et.SPIClient()
    # else:
    # port = args.serial_port or et.utils.autodetect_serial_port()
    port = et.utils.autodetect_serial_port()
    client = et.UARTClient(port)

    client.squeeze = False

    sensor_config = et.configs.IQServiceConfig()
    # sensor_config.sensor = args.sensors
    sensor_config.range_interval = [0.2, 1.0]
    ...

```

kmoretina, 2022-01-20:
Thank you for the information. I tried this and kept getting an error below

kmoretina, 2022-01-20:
Thank you for the information. I tried doing this, but I get an error shown below:

Multiple serial ports are available:
  COM4           (USB Serial Port (COM4))
  COM3           (Intel(R) Active Management Technology - SOL (COM3))
  COM1           (Communications Port (COM1))

Re-run the script with a given port

Is there any way to get around this that you know of?

andrehedesand, 2022-01-21:
Hi again @kmoretina,

Yes, this happens because ` et.utils.autodetect_serial_port()` was not absolutely sure which port to pick.

You can get around this by setting it yourself:

```
    ...
    #    client = et.SPIClient()
    # else:
    #     port = args.serial_port or et.utils.autodetect_serial_port()
    port = "COM<X>"  # port = "COM4"
    client = et.UARTClient(port)

    client.squeeze = False
    ...
```

Where `COM<X>` is the port of the sensor. For me, it was the port with the name "*USB Serial Port*", so `COM4` in in your case. If that does not work, try the other ports that is listed or get back to me! 

Cheers!

closed 2022-02-14 by AndersBuhl