acconeer/acconeer-python-exploration #103: issues about XM111 module's function

Undertaker7533967, 2022-01-24:
Hallo~~Sir
This is Riko from Taiwan 
We are studying in XM111 Module combine with STM32CUBEIDE.
We have a lot of  sample codes include tank_level , rf_certification, ....
We want to know if there is obstacle detection inside??
We have been searched obstacle detection.py also from Github.
Could you provide us obstacle detection.c after then we are kindly pleasure for it . 

all the best~~
By King

vackraetraed, 2022-01-24:
Hello King/Riko,

The obstacle detection is implemented in source code in the python example.
For our C SDK we have a detector library which you can use to run the obstacle detector.
libacc_detector_obstacle.a, there is a corresponding example file in our SDK called "example_detector_obstacle.c"

Undertaker7533967, 2022-01-25:
Dear Anton Martinsen:

 

Thanks for your information about  example_detector_obstacle.c 

As attached file is our final researching website 

we didn’t find libacc_detector_obstacle.a, and corresponding example_detector_obstacle.c for Acconeer SDK by any way

So please provide us the "example_detector_obstacle.c" or where we could download the it

 

All the best~~

Undertaker7533967, 2022-01-25:
Dear Anton Martinsen:

 

We have found out the libacc_detector_obstacle.a , it seems like package file which is lib and couldn’t run after implement it.

As attached is our researching results pictures .

So please provide us the detail addresses to download  "example_detector_obstacle.c"

Thanks a lot 

 

All the best~~

vackraetraed, 2022-01-25:
Hello King,

Which module exactly are you using? You stated XM111, however we have no module named this for sale.
Where did you find the libacc_detector_obstacle.a file?
If you tried to attach any pictures, I can not see them here.
What error do you get when you try to run example_detector_obstacle.c together with the libacc_detector_obstacle.a library?

Undertaker7533967, 2022-01-26:
Dear Anton Martinsen:

 

Sorry confused on you~~

We are suing Sparkfun A111 Pulsed Radar module:

 

We finally found out the example_detector_obstacle.c and libacc_detector_obstacle.a in CORTEXT M7 from Software downloads .

We have include acc_detector_obstacle.h，acc_detector_obstacle_configuration.h，acc_detector_obstacle_processing.h  

We have include 5 .a file into lib Sequentially as below:

libacc_detector_distance.a

libacc_detector_obstacle.a

libacc_detector_presence.a

libacc_rf_certification_test.a

libacconeer.a

 

As attached video is you mention before:

 

https://drive.google.com/file/d/1CuP1phZB11verSHPNvbee2E16eggJY0Q/view?usp=sharing

 

The error is undefined acc_detector_obstacle_create，acc_detector_obstacle_activate，acc_detector_obstacle_destroy……

So our question is M7 source code could be compiled on stm32cubeIDE??

Or what’s steps is wrong and need to modified??

 

All the best~~

Undertaker7533967, 2022-01-28:
Dear Anton Martinsen:

 

We have another issues about CORTEX M4 source could be compile with Microchip  MCU MPLAB or Studio .

Due to Microchip MCU is in stock and getting it convenient this year.

We have been tested CORTEX M4 source but couldn’t compile it successfully after following STMCubeIDE setting processes as attached file.

 

So We would like to know if which software as below could be compiled with Microchip MCU MPLAB or Studio:

acconeer_cortex_m4_armcc_v2_10_0

acconeer_cortex_m4_armclang_v2_10_0

cortex_m4_armclang

cortex_m4_iar

nrf52_module_software

 

all the best~~

 

 

 

 

 

https://drive.google.com/file/d/1CuP1phZB11verSHPNvbee2E16eggJY0Q/view?usp=sharing

vackraetraed, 2022-01-28:
Hello @Undertaker7533967 

When using STM32CubeIDE (with the arm-gcc toolchain) you need to specify the libraries to the linker.
See STM32CubeIDE guide chapter 3.1.3 on how to add libraries in the IDE.
What MCU are you using currently? The Cortex-M7 libraries might not work unless you have a Cortex-M7 processor.

If you are using a Cortex-M4 you can try and download the acconeer_nrf52_module_software, this software have the Obstacle files available for Cortex-M4 even though we do not recommend using it since Obstacle requires quite a lot of processing power.

Which firmware package you download depends on which toolchain you will be using.
My guess however is that you will be using the arm-gcc toolchain.
Then the libraries in acconeer_cortex_m4 and nrf52_module_software should work since they have been compiled using arm-gcc for a Cortex-M4 processor.

Undertaker7533967, 2022-02-07:
Dear Anton Martinsen:

 

Thanks for your explanations about Obstacle detection issues .

As we know XM112 PCR Module is using Microchip MCU Arm M7 ATSAME70Q20A-CFN as attached file 

And because of STM MCU is out of storage, so we want to start the PCR sensor with  Microchip MCU Arm M4 ATSAMD51G19A.

1.     As you mention that could it be compile ATSAMD51G19A with acconeer_cortex_m4 and nrf52_module_software??

2.     Are any relationship with Microchip user guide like STM32CubeIDE guide chapter 3.1.3??

3.     What are processing power using obstacle detection?? Could you explain more detail information to design obstacle detection sensor?? 

 

All the best~~

vackraetraed, 2022-02-09:
Hello @Undertaker7533967 .

Well, a Cortex-M7 is preferred for any Obstacle detection use case but it can maybe be done on a Cortex-M4, since the Obstacle detector requires a lot of processing.

1. Yes, the libraries in acconeer_cortex_m4 and nrf52_module_software should be able to run on ATSAMD51G19A as long as the arm-gcc toolchain is used.
2. There is no guide for Microchip products. But the guide can be used as a general guide on what you want to accomplish. The things you want to do are generally the same they are just done a bit differently between different companies IDEs and develepoment tools.
3. The obstacle detector library just requires more processing of the data. Hence a faster processor with an FPU is needed to run it more efficiently.

Undertaker7533967, 2022-02-15:
Dear Anton Martinsen:

 

We try to use Microchip AtemelStudio 7 Studio compiler  and running source code as same as folloing STM32 IDE guide but fail a lot:

If running STM32CubeIDE then this can be solved by going into "Project" -> "Properties" -> "C/C++ Build" -> "Settings" -> "MCU GCC Linker" -> "Libraries" and adding "acc_detector_distance" to the "Libraries" field. Notice that the detectors needs to be linked before the "acconeer" library and need to be above it. Make sure that the "acconeer" library has been added as well.

The question is how to add several .a file to libraries on Microchip AtemelStudio 7 Studio compiler??

Could you assist us the user guide for crate a folder and import .c .h and .a  files and compile successful??

 

All the best~~

vackraetraed, 2022-02-15:
Hello @Undertaker7533967 

I would suggest looking at Microchips support for advice on using their IDE.
However, when googling I found this:
https://microchipsupport.force.com/s/article/Atmel-Studio--Creating-and-linking-GCC-C-Static-Library-Project

It seems to add the libraries to the linker in Atmel Studio:
1. Navigate to Solution Explorer Window.
2. Right click on the project name and select ‘Add Library'.
3. Select ‘Browse Libraries’ on the left side pane.
4. Click on ‘Browse’ button.
5. Navigate to the folder containing the acconeer libraries.
6. Select the library file ‘xxxx.a’

Best regards,
Anton

closed 2022-02-23 by vackraetraed