acconeer/acconeer-python-exploration #106: Eliminating Buffer

kmoretina, 2022-02-15:
I am using the IQ service for a project. I was wanting to sample data at a slower rate than what the IQ service samples at. I noticed in client.py there was a buffer and was wondering if there was a way to get rid of this buffer? I also know I am needing a sample every 100 milliseconds if there is a known way to find data every 100 milliseconds and it would be easier than eliminating the buffer. Thank you

AndersBuhl, 2022-02-16:
Hi,

If you want to get data every 100 millisecond, you should limit the update rate to 10 Hz. This is correct way to do this.

BR,
Anders

kmoretina, 2022-02-22:
Is there a way to eliminate the buffer? I tried changing the update rate, and I'm still having the same issue for my application.

AndersBuhl, 2022-02-25:
Hi,

I'm not sure if I understand what issue you are having. If you set the update rate to 10 Hz, you will get data every 100 ms.

Could you please in more detail describe your application and the issues you are having?

What buffer are your referring to?

BR,
Anders

kmoretina, 2022-03-03:
There is a buffer in the client.py file when the code calls the function client.get_next() and I was having issues getting accurate data. I am integrating the xm112 with different software calling client.get_next() at a different rate than the standard configuration called for. I got it figured out though. Thank you!

While this issue is open, I was curious why the minimum depth range is 6 cm? Is there anything in the hardware limiting this, and if not, could I set the range lower than 6 cm?

AndersBuhl, 2022-03-04:
The is no hardware limit, but the service is limited because it uses information in the range 0-6 cm to calibrate the sweep. The envelope service has the possibility to get closer.

closed 2022-03-23 by AndersBuhl