acconeer/acconeer-python-exploration #108: XM112 radar software examples

ArtsiomHalash, 2022-04-26:
Hello, I am looking for some help with running pythone examples.

I am currently trying to run python examples on XM112 with XB112. I use VBox Ubuntu machine. 

The image below shows an example of sparse_speed python script, after initiation it freezes in the same state. 

[image]

This is sparse.py services example. I run 
```
python3 sparse.py -vv -u /dev/ttyUSB0
```
and get output below. 
```
10:42:17.444 | DEBUG   | MainProcess      | acconeer.exptool.a111._clients.reg.client | got invalid frame (incorrect end marker), attempting recovery

10:42:17.455 | WARNING | MainProcess      | acconeer.exptool.a111._clients.reg.client | successfully recovered from corrupt frame

raise ClientError("got invalid frame and could not recover")

acconeer.exptool.a111._clients.base.ClientError: got invalid frame and could not
```
Is there an issue with configuration settings? What direction should I look towards?

AndersBuhl, 2022-04-26:
Hi ArtsiomHalash,

First of all it states that you should upgrade your client. So try to this as one of the steps.

You could try to flash the XM112 with the latest available module software.

Is it possible for you to run it on a native Ubuntu or Windows, to remove the VBox layer?

BR,
Anders

ArtsiomHalash, 2022-04-26:
> Hi Anders,
Thank you for quick response. 

I think 2.11.1 is the latest version, could you suggest a way for updating the client?
I use the latest firmware.

I run it on windows first, the GUI works well but most of the python scripts give packages errors, after trying to solve them for a while I decided to use vm ubuntu instead.

AndersBuhl, 2022-04-26:
Correct 2.11.1 is the latest version.

to upgrade you should run 'python -m pip install -U acconeer-exptool[app]'

It should be possible to run the standalone examples on Windows, if acconeer-exptool and all dependencies is correctly installed, which they should be if you run the command above.

ArtsiomHalash, 2022-04-26:
I reinstalled and used the upgrade on the exptool but as you can see I still get please upgrade client error. 
It solved the packages problem for now, I can run services scripts. 

When I run speed_sparse script I get this:
[image]

```
RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
```

AndersBuhl, 2022-04-27:
Hi,

The SDK version in exploration tool have not been updated. This will be done during today. That's the reason for the version warning.

To run standalone examples use the following command: 'python -m acconeer.exptool.a111.algo.speed_sparse -u COM3'

ArtsiomHalash, 2022-04-27:
This error appears when I run speed_sparse, most other examples work but I need to measure the speed for my project

vackraetraed, 2022-04-28:
Hello @ArtsiomHalash ,

We have acknowledge that this bug seem to happen on Windows when running the example standalone. (looks like it still works when running from the GUI)
We will file a bug on it.

If you really have to run this specific example in the near future it is probably best to run it on native Ubuntu.

ArtsiomHalash, 2022-04-28:
Hi @vackraetraed ,

I see, thank you very much for the information. Could I trouble you with a few more questions.. 

I'm quite new to this, would appreciate a few suggestions, how to start writing my own detector and when standard module software can and cannot be used? 

Can xm112 and xb112 be sued with Raspberry pi ? 

Also how is the output used in a different application? For examples, obstacle detection on a robot

ArtsiomHalash, 2022-04-28:
Also I am interested in exploring Radio Frequency signature classification with ML, would it be possible with A111?

AndersBuhl, 2022-05-02:
Hi,

For adding your own detector, you can follow the guide here: [Adding your own algorithm module](https://docs.acconeer.com/en/latest/gui/working_with_the_gui/adding_your_own_algorithm_module.html)

It is possible to run the XM112 from a Raspberry PI. It is a similar setup as for Ubuntu.

It depends a lot on the application how the output is utilized. As for obstacle detection, one could be have the robot move in an opposite direction when detecting that an object is getting too close.

What kind of classification are you thinking of?

ArtsiomHalash, 2022-05-03:
I want to classify RF signatures of different objects like vehicle, cyclist or a pedestrian for example, basically to have a radar system capable of differentiating types of object on the road.

AndersBuhl, 2022-05-04:
Hi,

This would be very difficult to do. The radar would have to be able to pick up different characteristics for the different objects, e.g. speed, size or movements. A classifier would then have be accurate enough to the differentiate characteristics from each other. Using multiple sensors at different positions and angles might make it easier to pick up essential characteristics.

ArtsiomHalash, 2022-05-09:
Hello,

thanks for the info, I'll have to explore more on this topic.

I wonder if there is a ready speed detector example for xc112/xr112 ? I couldn't find one in the files. 

The xm112 works with rpi but I have issues with Qt6 libraries, I couldn't install pyside6 to rpi3. Would it work if I get rid of the gui code and just get simple output? Also how can I run the detector examples from "algo" directory, like the "speed_sparse"?

AndersBuhl, 2022-05-09:
Hi,

The speed detector example you're referring to is the speed_sparse which is located in the algo lib (src/acconeer/exptool/a111/algo/speed_sparse)

Yes, it is possible to omit the GUI part. You do this by installing the algo package instead of the full app, using this command 'python3 -m pip install acconeer-exptool[**algo**]'

To run standalone examples: 'python -m acconeer.exptool.a111.algo.speed_sparse -u COM3'
To be able to run it without GUI you can modify the main function to skip everything referring to 'pg_process' and instead print the processed data (or something similar)

ArtsiomHalash, 2022-05-10:
Brilliant, thank you.

Does speed_sparse exist in C for XC/XR112 ? I described the issue with the bug above on XM112,  speed_sparse won't run on windows. I try to run it on RPI but I couldn't install python QT6 libraries, I will take out the interface code to run it. Is there  a standalone speed_sparse for RPI?

AndersBuhl, 2022-05-11:
Hi,

There are no speed_sparse in C, the implementation is a python example only. The bugs in exploration tool is not dependent on HW, but rather how Windows and PyQT handles threads. The standalone speed_sparse is the same regardless of the host HW. Running exploration tool on RPi is possible, but the GUI is difficult to run. The easiest way to do it is to look at src/acconeer/exptool/a111/algo/_standalone_main.py and write your own main to exclude all GUI parts.

ArtsiomHalash, 2022-05-25:
Hey guys,

I'm working with presence detector now on XM112, I set it up on RPI4 to cover certain are in the office and detect presence there. Is it possible to determine an approximate amount of people or at least how crowded the area is? 
When we run Sparse service in exploration GUI, movement history graph shows multiple patterns if there are several people present. My idea was to take an array of sweeps and focus on the peaks where the value surpasses threshold, so the more peaks the more area is crowded.

closed 2022-06-21 by AndersBuhl