acconeer/acconeer-python-exploration #110: Envelope 3d Demo

Ryanstak, 2022-05-08:
[image]

This image is in the introduction section and it states that this is the output of the envelope service in the exploration tool. Mine doesn't look like that. Am I missing a setting or something?
[image]

AndersBuhl, 2022-05-09:
Hi,

The 3D version of envelope was removed in 4.0.0. It is still available in the Radar Viewer if you checkout the v3-branch.

AndersBuhl, 2022-05-10:
Hi,

My mistake, it was removed earlier. This is the last commit containing the radar_viewer 1d4069bf71d21cde91ffd1e6074f73df3258f0d8. Please checkout that and find the radar viewer folder in the root.

Ryanstak, 2022-05-14:
Thank you. Is there anyway to get the metadata from the 3d demo, or is it just a live view?

Thanks,

Ryan Stakenborghs
Stak Information Consultation

AndersBuhl, 2022-05-17:
Hi,

It is only meant for a live view. To get more in-depth data, please refer to the exploration tool GUI and example scripts.

Ryanstak, 2022-05-23:
If I wanted to create a module that displayed a graph similar to the distance detector, but in 3d. Would pyqtgraph or matplotlib be better for this? It looks like both have the capability.

closed 2022-06-21 by AndersBuhl