acconeer/acconeer-python-exploration #111: Help for connecting XM132 Module

rohithanaydy, 2022-05-17:
Hello, This is not a bug. This is a problem I am facing. I am trying to use the Exploration tool with XM132 Module alone. I am not using XE132 EVK.  ` acc module server.bin ` is flashed on the module using stm32 utility link and connected to the Exploration Tool using USB to UART converter with BaudRate = 115200, and Hardware Flow control enabled.
But it shows  **Could not connect to the server** . What will be the reason ? Anyone please tell me to solve the issue

Thanks

vackraetraed, 2022-05-18:
Hello @rohithanaydy 

I created a new binary for you. Let me know whether or not it works and we can take it from there:
[acc_module_server.zip](https://github.com/acconeer/acconeer-python-exploration/files/8714293/acc_module_server.zip)

vackraetraed, 2022-05-18:
@rohithanaydy 

Also, which version of the exp tool are you running?
And which version of pyserial do you have installed?

vackraetraed, 2022-05-18:
Hello again @rohithanaydy 

Considering the pyserial code seems to call Windows DLL libraries, I am now kind of suspecting that there are drivers missing for the USB to UART converter.
What type of USB to UART converter are you using? What brand?
For example if it is an FTDI then perhaps this link would be good: https://ftdichip.com/drivers/vcp-drivers/

rohithanaydy, 2022-05-18:
Dear @vackraetraed 
 Thank for your reply. I cant load the  `module_software_example.py` using stm32 st link utility. 
The USB to UART converter I am using is Silicon Labs CP210 USB to UART Bridge.
I downloaded the exploration tool  with reference to    [](https://github.com/acconeer/acconeer-python-exploration#quickstart-for-windows) . The version is 4.2.0 . 
Thank you
Rohith R

vackraetraed, 2022-05-18:
Hello @rohithanaydy 

module_software_example.py is a python script you run on your PC that communicates over serial with the module software flashed on XM132.
But if you are using Silicon labs CP210 USB to UART Bridge then you probably need to install the VCP drivers that can be found here: https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers
Have you tried installing these drivers?
If you go into device manager, does Silicon labs CP210 USB to UART show up as a Virtual COM port when you connect the USB?

vackraetraed, 2022-05-18:
Hello @rohithanaydy 

Yes, pyserial is needed to run Exploration Tool since it communicates by serial in python.
The firmware you downloaded does not seem to be quite correct.
My guess is that you would need to install this: https://www.silabs.com/documents/public/software/CP210x_Windows_Drivers.zip

rohithanaydy, 2022-05-19:
Dear @vackraetraed 
Installed the driver CP210xVCPInstaller_x64, py serial- 3.5 , keeping the UART Config same, But still the device is not connected to Exploration Tool.
I am using windows 11, is there any problem ?

vackraetraed, 2022-05-20:
Hello @rohithanaydy 

I will close this issue on github, since you have an open ticket in our freshdesk regarding the same issue.
I have provided Ola with the information given in the comments here.

closed 2022-05-20 by vackraetraed