acconeer/acconeer-python-exploration #112: Exploration Tool on Raspberry Pi with PySide 6

chadrockey, 2022-05-26:
The old exploration tool ran great on the same raspberry pi running the streaming server.

It seems that qt6 and pyside6 aren’t yet supported on raspberry pi.  Is there any alternative?

Is running the streaming server on the raspberry pi a feature you would like to preserve?  Or was that ability mostly unintended?

chadrockey, 2022-05-26:
Also, does the main readme need to be adjusted?  It seems like PySide6 is also not supported for Ubuntu 18.04

AndersBuhl, 2022-06-07:
Hi,

Running the GUI on a raspberry is not possible, but you can install and run the algo parts and run through python scripts.

AndersBuhl, 2022-06-21:
Hi,

Thank you for your input!

We will update the instructions

closed 2022-06-21 by AndersBuhl