acconeer/acconeer-python-exploration #115: Missing breathing examples

kmtm, 2022-07-07:
**Describe the bug**
The latest version of the repo is missing the 'processing' folder that has breathing.py and sleep_breathing.py examples. Downloading a a prior commit version that does have these files and running them produces the following error:

`AttributeError: module 'acconeer.exptool.utils' has no attribute 'ExampleArgumentParser'`

Are there updated version of these examples coming soon? I need a bit of help understanding how to get the sleep breathing algorithm (https://docs.acconeer.com/en/latest/exploration_tool/algo/a111/sleep_breathing.html) up and running to test out. 

**To reproduce**
Steps to reproduce the behavior:
1. Download this commit: https://github.com/acconeer/acconeer-python-exploration/blob/73d06aa5175a205747a0a84999a94401a223e90a/examples/processing/breathing.py#L228
2. run breathing.py or sleep_breathing.py

**Environment (please complete the following information):**
 - Host OS: Raspbian on Raspi 4
 - Acconeer hardware: XM112 and breakout board over serial USB
 - Acconeer software version: module firmware v2.11.1
 - Exploration version: 5.0.0

AndersBuhl, 2022-07-08:
Hi,

The example processing have been moved. The two you are referring to are now moved to: src/acconeer/exptool/a111/algo/breathing and src/acconeer/exptool/a111/algo/sleep_breathing. These folders each contain a _processor.py which includes all code used for processing.

The examples can be run by: 'python -m acconeer.exptool.a111.algo.breathing' and 'python -m acconeer.exptool.a111.algo.sleep_breathing'

BR,
Anders

kmtm, 2022-07-12:
Many thanks! I'm now dealing with the PySide6 requirement and the fact that Raspi only seems to support PySide5 at the moment. Any suggestions there?

kmtm, 2022-07-12:
[UPDATE] In case anyone is working with a Raspi 4, I was able to get this to work by using the v3 version of the acconeer tool (following the directions for manually building, though there were a few pip errors on versions I ignored) with PyQT5 installed. Make sure to pip uninstall the latest acconeer if you had tried that first, though, otherwise it will be angry :)

AndersBuhl, 2022-07-12:
Hi,

The issue with PySide6 on Raspberry have been discussed earlier: [Exploration Tool on Raspberry Pi with PySide 6 #112](https://github.com/acconeer/acconeer-python-exploration/issues/112)

BR,
Anders

kmtm, 2022-07-12:
Yeah, I looked into that. It seems like you can't get the UI in the latest exp tool version like you can in v3.

AndersBuhl, 2022-07-13:
That is correct. It has never been officially supported and there have been some issues to get the GUI running on a Raspberry PI, but it have been doable before.

closed 2022-07-13 by kmtm