acconeer/acconeer-python-exploration #116: basic.py understanding

kmtm, 2022-07-13:
Hi -- not a bug, but rather a question. 

I've read through much of the online documentation and code comments, but I'm still trying to get a grasp on the basics of this sensor given that I am not an EE or practiced in signal processing. Could you help me understand:

1. What type of distance data is the sensor (A111) getting - a single point? a line? a plane? (and if more than a point, is this accomplished through sweeping?) Does this vary in the Envelope vs. IQ service? 
2. What does a single sweep in basic.py represent? (ex. is the sweep horizontal vs. vertical? what do each of the 207 printed data points per sweep represent?)
4. In the most basic GUID example showing the envelope graph for a cup (https://docs.acconeer.com/en/latest/exploration_tool/gui/working_with_the_gui/collect_data.html), what does the amplitude truly represent? For instance, is the tip of the peak the model's best guess of the distance to the closest edge of the cup (and thus the rise and fall on either side is just built in uncertainty)?

I'd also be open to a beginner-friendly overview if you have it. Most of what I have found elsewhere (online videos, lectures, etc.) is either overly simplified or requires deeper spectral understanding.

AndersBuhl, 2022-07-14:
Hi,

To get an (somewhat extensive) introduction to the sensor there is a sensor introduction available on the [Developer site](https://developer.acconeer.com/)

1. The radar waves are transmitted in on oval cone form. The distance dimension of the data corresponds to a specific distance from the sensor in any direction (limited by the transmission angle in the E- and H-plane).
2. For sweep direction see answer above. Each of the point is an amplitude value of the received signal. High amplitude indicates a strong reflection.
3.  For more information on the Envelope service, read [this guide](https://docs.acconeer.com/en/latest/handbook/a111/services/envelope.html). The amplitude at each point corresponds to the amount of energy reflected at that point.

kmtm, 2022-07-14:
Gotcha, many thanks!

So for the 207 points/sweep in basic.py, does each point represent a different distance (within the bounds of the cone of the E-H planes) and the corresponding energy at that distance? Or, is each data point representing a specific E-H coordinate?

AndersBuhl, 2022-07-15:
Hi,

Each point is just a single value (not a coordinate) and corresponds to the returned energy at that point. An object straight in front of the sensor at 20 cm distance will show up on the same distance point as an object at a 45 degrees angle 20 cm from the sensor, since they are at the same distance from the sensor.

closed 2022-08-16 by AndersBuhl