acconeer/acconeer-python-exploration #118: Could not connect to server

DragosLeuBre, 2022-09-21:
Hello all,

Recently, I tried connecting to my XC112+XR112 devices (mounted on a Raspberry Pi 4), through the exploration server. A quick note, in the past I was able to connect without any issues, but with this SW version I have the following issue: once I try to connect to the specified IP address, it pops up the "Could not connect to server" message.

From Putty I get the following messages:

01:22:35.752 (I) (exploration_server_base) Got command: get_version
01:22:35.752 (E) (exploration_server_base) Unknown command get_version.
01:22:35.752 (E) (exploration_server_base) Sending error "Unknown command" to client
01:22:35.760 (I) (exploration_server_base) Got command: get_system_info

While from the GUI I get:

Traceback (most recent call last):
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\app\old\app.py", line 1610, in connect_to_server
    self.client.start_session(config)
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\client.py", line 111, in start_session
    return self.subclient.start_session(config, check_config)
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\base.py", line 114, in start_session
    ret = self.setup_session(config, check_config=check_config)
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\base.py", line 89, in setup_session
    session_info = self._setup_session(config)
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\json\client.py", line 543, in _setup_session
    info = self._init_session()
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\json\client.py", line 569, in _init_session
    return self._protocol.init_session()
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\json\client.py", line 388, in init_session
    self._send_cmd(self._session_cmd)
  File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111\_clients\json\client.py", line 36, in _send_cmd
    s = json.dumps(cmd_dict, separators=(",", ":"))
  File "json\__init__.py", line 234, in dumps
  File "json\encoder.py", line 199, in encode
  File "json\encoder.py", line 257, in iterencode
  File "json\encoder.py", line 179, in default
TypeError: Object of type Profile is not JSON serializable

Has anyone any idea on how to fix this?

Thank you!

DragosLeuBre, 2022-09-21:
Hello @vackraetraed

The version is Lib 5.2.4

Kind regards,
Dragos

vackraetraed, 2022-09-21:
Also, which firmware version of the exploration server is running on the Raspberry PI?

Best regards,
Anton

DragosLeuBre, 2022-09-21:
It's acconeer_rpi_xc112_a111-v2_13_0, which should be the latest one (as I downloaded it this morning from the site). With the previous (v2_12) I had no issues (also, with a previous version of the exptool, which I don't remember the versione unfortunately).

Generally, I should be using the latest versions of the software/firmware.

Kind regards,
Dragos

vackraetraed, 2022-09-21:
I'd try to purge the configs before running:
python -m acconeer.exptool.app --purge-config

Though I have never tried running Windows or the portable tool myself.
Perhaps the command needs to be inserted at some specific part of the start up.

Best regards,
Anton

vackraetraed, 2022-09-21:
You can also try to print the cmd_dict in File "C:\Users\conAlt41\Desktop\Python\portable_exploration_tool\tools\python-3.9.10-embed-amd64\Lib\site-packages\acconeer\exptool\a111_clients\json\client.py", line 36, in send_cmd
s = json.dumps(cmd_dict, separators=(",", ":"))

And post its result here.

Best regards,
Anton

DragosLeuBre, 2022-09-21:
Thank you for the suggestions, I tried both without any results. 

This is what gets printed:

{'cmd': 'get_version', 'api_version': 3}
15:53:13.780 | INFO    | MainProcess      | acconeer.exptool.a111._clients.json.client | No get_version cmd, try get_system_info
{'cmd': 'get_system_info', 'api_version': 3}
15:53:13.789 | INFO    | MainProcess      | acconeer.exptool.a111._clients.base  | reported version: 2.13.0

Kind regards,
Dragos

vackraetraed, 2022-09-21:
Strange, is the cmd_dict empty?
You did not try to print it after the failing command?

This is what it looks like for me:
`{'cmd': 'get_version', 'api_version': 3}
16:17:45.079 | INFO | MainProcess | acconeer.exptool.a111._clients.json.client | No get_version cmd, try get_system_info
{'cmd': 'get_system_info', 'api_version': 3}
16:17:45.085 | INFO | MainProcess | acconeer.exptool.a111._clients.base | reported version: 2.13.0
{'cmd': 'setup', 'repetition_mode': 'on_demand', 'groups': [[{'sensor_id': 1, 'service': 'sparse', 'config': {'range_start': 0.18, 'range_length': 0.6000000000000001, 'gain': 0.5, 'hw_accelerated_average_samples': 10, 'profile': 2, 'downsampling_factor': 1, 'maximize_signal_attenuation': 0, 'sampling_mode': 1, 'sweeps_per_frame': 16, 'tx_disable': 0, 'power_save_mode': 'active', 'asynchronous_measurement': 1, 'mur': 6}}]], 'api_version': 3}`

vackraetraed, 2022-09-22:
Hi @DragosLeuBre 

I managed to reproduce the issue on Windows 11, are you using Windows 11?

vackraetraed, 2022-09-22:
Hello @DragosLeuBre 

Sounds strange, I am getting the exact looking error as you had before but my command that is being sent is for sure looking quite strange here. I will continue looking into it, maybe it is related.
Specifically the profile field looks like this: 'profile': <Profile.PROFILE_2: ('2', 2, 0.12)>
It should look like 'profile': 2
So in this case it is quite logical that it would produce the error which both you and me see.

What error are you facing when using previous versions?

DragosLeuBre, 2022-09-22:
Thank you @vackraetraed 

It's not an error per se, it's just an incompatibility between the sensor firmware and the exptool client versions, as they don't match, and as soon as I run the client I receive a fail saying that I need to update the client.

vackraetraed, 2022-09-23:
Hello again @DragosLeuBre 

It looks like Windows does not recognize enum as enum.
Temp fix is to go into acconeer/exptool/a111/_clients/json/client.py
add "from acconeer.exptool.a111._configs import ConfigEnum" in the beginning of the file.

Then in the JsonProtocolExplorationServer class in the same file.
In function setup_session, there is a check:
elif isinstance(config_val, enum.Enum):
Change this to:
elif isinstance(config_val, ConfigEnum):

I could at least start the Exploration tool with envelope using this.
We will work on updating the exploration tool fully with this.

DragosLeuBre, 2022-09-23:
Hello @vackraetraed 
I can confirm that now it works with the tweaks mentioned.

Thank you!

vackraetraed, 2022-09-23:
Hello @DragosLeuBre 

Excellent!
From further analysis it seems this problem came from installing shiboken6==6.3.2 (released last monday)
shiboken6 somehow now imports its own definition of enum which does not equal the standard python enum.
Revering shiboken6 to shiboken6==6.3.1 should also fix the issue without altering the code.

closed 2022-09-28 by vackraetraed