acconeer/acconeer-python-exploration #120: Could Not Open Port : PermissionError 'Access is denied'

JCJC7, 2022-11-03:
Hi, I've recently purchased the Acconeer XE132 kit and trying some of its functions. I have tried with the GUI to get the basic idea of the radar services. I also successfully run the python scripts for the basic services as guided in the github page (except IQ services as it shows that IQ mode is not supported). However, when I try to run the python code for **sparse_fft** which I downloaded from github, it pops out this error. Here I would like to seek for solution to this error in the below figure. Thank you in advance. 

[image]

stijnb1234, 2022-11-03:
_(Your Windows username is still visible)_

Make sure the module is connected to COM8 and that COM8 is not used by anything else.

JCJC7, 2022-11-04:
Hi, thank you for the reply.
Yes, I'm sure that the module is connected to COM8. This is the figure for envelope python script. 
[image]

If I run the **sparse_fft** __main__.py script, the same error still occurs.
I tried with other algorithm python scripts, it is all showing the same error. Did I do something wrong or I need to do something before running these scripts?  

path : ...\github-acconeer-python-exploration-master\src\acconeer\exptool\a111\algo
[image]

vackraetraed, 2022-11-04:
Hello,

It seems to work fine for me to run these algo examples on Linux, however, when running the same code on Windows I get the same error as you. 
So might be some Windows specific permission that is not granted when running the algorithms standalone like this.
Perhaps there is some clue in the differences when establishing a UART connection in the standalone examples vs the standalone algo ones.

vackraetraed, 2022-11-07:
Could you try running the firmware from the algo map with the following command:
python -m sparse_fft -u COM8

vackraetraed, 2022-11-07:
Perhaps I meant the sparse_fft map :D

vackraetraed, 2022-11-07:
Sorry, I questioned myself too much.
You should write the command in the algo folder, I was correct the first time.

vackraetraed, 2022-11-07:
You can also try:
python -m acconeer.exptool.a111.algo.sparse_fft -u COM8

From any folder if you have installed the package.

JCJC7, 2022-11-07:
Thank you for your support! The final command works!. 

[image]

Truly appreciate your time and effort. Thank you!

closed 2022-11-07 by JCJC7