acconeer/acconeer-python-exploration #122: the issue of setting up XC120+XE121 module

Undertaker7533967, 2023-01-03:
**Describe the bug**
We couldn’t find out module as attached picture (USB device on Riko side):
[image]

it shown XC 120 bootloader and also shown DFU not found on CMD surrounding (DFU not found):
[image]

Instead of XC 120 Exploration Server as attached picture on acconeer website (USB device on acconeer side):
[image]

We don’t know how to solution it 
Could you please assists us ??

all the best~~

AndersBuhl, 2023-01-03:
Hi,

To resolve this you need to use Zadig. If Zadig states that you are using XC120 Bootloader the driver version should be 'usbser'. Then try to flash it.

Undertaker7533967, 2023-01-06:
Dear Anders:

Thanks for your answer, we have already finished it, it is related with driver USB and need to be reflash.  we suggest creator could be adding into github of setting up XC120 module processes.

all the best~~