acconeer/acconeer-python-exploration #124: Com port issue with exploration tool Xm112/Xb112

nehabhootesh, 2023-04-05:
Hello team, we have started working with xm112 and xb112 for our mini project. We followed all steps as in the youtube channel (acconeer official yt channel), but after starting exploration tool on windows 10 ad ubantu we are facing the issue with the com port. we checked with the device manager and port is clearly mentioned as com3. Ubantu we used command prompt and it showed as usb0. But lately both the exploration tools are not connecting to the server.
Here is the screenshot of both windows and ubantu
[image]
[image]
we have used the latest exploration tool software available in the official website.
Versions used:
windows 10
python 3.10.6
exploration tool: latest using command prompt
Bossa version: bossa-x64-1.9.1.msi
Module SW for XM112: 2.14.2

We are waiting for someone to help us out earliest!!
Thanks and regards
Neha B

nehabhootesh, 2023-04-07:
Hey can someone look into our issue and resolve it earliest!
Thanks 
Neha B

nehabhootesh, 2023-04-10:
python3 -m [acconeer.exptool.app](http://acconeer.exptool.app/)
qt.dbus.integration: Could not connect "org.freedesktop.IBus" to globalEngineChanged(QString)
16:05:24.778 | INFO    | MainProcess      | [acconeer.exptool.app.old.app](http://acconeer.exptool.app.old.app/)         | Loaded configuration from last session: /home/sitec/.local/share/acconeer_exptool/last_config.npy
16:05:33.146 | WARNING | MainProcess      | [acconeer.exptool.app.old.app](http://acconeer.exptool.app.old.app/)         | Overriding baudrate (115200)!
16:05:33.248 | INFO    | MainProcess      | acconeer.exptool.a111._clients.base  | reported version: 2.14.2
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 1169, in update_canvas
    new_canvas = self.init_graphs()
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 298, in init_graphs
    self.reload_pg_updater(canvas=canvas)
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 322, in reload_pg_updater
    self.service_widget.setup([canvas.ci](http://canvas.ci/))
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/a111/algo/envelope/ui.py", line 27, in setup
    self.ampl_plot = win.addPlot(row=0, colspan=num_sensors)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/GraphicsLayout.py", line 72, in addPlot
    plot = PlotItem(**kargs)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/PlotItem/PlotItem.py", line 154, in __init__
    self.titleLabel = LabelItem('', size='11pt', parent=self)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/LabelItem.py", line 19, in __init__
    GraphicsWidget.__init__(self, parent)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/GraphicsWidget.py", line 18, in __init__
    QtWidgets.QGraphicsWidget.__init__(self, *args, **kwargs)
TypeError: GraphicsWidgetAnchor.__init__() takes 1 positional argument but 2 were given
Traceback (most recent call last):
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 664, in start_btn_clicked
    self.start_scan()
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 1303, in start_scan
    self.update_canvas(force_update=True)
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 1169, in update_canvas
    new_canvas = self.init_graphs()
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 298, in init_graphs
    self.reload_pg_updater(canvas=canvas)
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 322, in reload_pg_updater
    self.service_widget.setup([canvas.ci](http://canvas.ci/))
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/a111/algo/envelope/ui.py", line 27, in setup
    self.ampl_plot = win.addPlot(row=0, colspan=num_sensors)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/GraphicsLayout.py", line 72, in addPlot
    plot = PlotItem(**kargs)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/PlotItem/PlotItem.py", line 154, in __init__
    self.titleLabel = LabelItem('', size='11pt', parent=self)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/LabelItem.py", line 19, in __init__
    GraphicsWidget.__init__(self, parent)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/GraphicsWidget.py", line 18, in __init__
    QtWidgets.QGraphicsWidget.__init__(self, *args, **kwargs)
TypeError: GraphicsWidgetAnchor.__init__() takes 1 positional argument but 2 were given
Traceback (most recent call last):
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 664, in start_btn_clicked
    self.start_scan()
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 1303, in start_scan
    self.update_canvas(force_update=True)
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 1169, in update_canvas
    new_canvas = self.init_graphs()
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 298, in init_graphs
    self.reload_pg_updater(canvas=canvas)
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/app/old/app.py", line 322, in reload_pg_updater
    self.service_widget.setup([canvas.ci](http://canvas.ci/))
  File "/home/sitec/.local/lib/python3.10/site-packages/acconeer/exptool/a111/algo/envelope/ui.py", line 27, in setup
    self.ampl_plot = win.addPlot(row=0, colspan=num_sensors)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/GraphicsLayout.py", line 72, in addPlot
    plot = PlotItem(**kargs)
  File "/home/sitec/.local/lib/python3.10/site-packages/pyqtgraph/graphicsItems/PlotItem/PlotItem.py", line 154, in __init__
[... cut ...]
TypeError: GraphicsWidgetAnchor.__init__() takes 1 positional argument but 2 were given
^C16:06:30.341 | INFO    | MainProcess      | [acconeer.exptool.app.old.app](http://acconeer.exptool.app.old.app/)         | Saved configuration from this session to /home/sitec/.local/share/acconeer_exptool/last_config.npy
16:06:30.364 | INFO    | MainProcess      | [acconeer.exptool.app.old.app](http://acconeer.exptool.app.old.app/)         | Saved configuration from this session to /home/sitec/.local/share/acconeer_exptool/last_config.npy
[image]
[image]

AndersBuhl, 2023-04-11:
Hi,
Always make sure that the exploration tool is updated by following instructions for your OS.
When your saying "ubantu" I assume you mean Ubuntu?
From the Windows screenshot it looks like it is connecting without issues, but when it comes to Ubuntu it does not.

Is your Ubuntu running in a virtual box inside Windows?

For you second problem:
I have not been able to reproduce this. Can you make sure that the version you are using is v6.0.4?

Also, can you provide the output from 'pip list'?

BR
Anders

nehabhootesh, 2023-04-11:
we are running on native ubuntu

nehabhootesh, 2023-04-12:
hey, we reduced the update rate still the frames are missed in IQ services.

AndersBuhl, 2023-04-13:
what is your exact configuration? can you send a screenshot of the settings?

nehabhootesh, 2023-04-16:
Hey, I wanna know can we directly calculate breathe rate and heart rate on breakout board itself, like by performing digital signal processing algorithm written in python language and send the heart rate and breathing rate into the ai model(rzv2ma board) using usb? If yes, let me know the steps to get started !!. If not then how to get the breath rate and heart rate like do we need to stream the raw data to any processing unit for processing?? please reply soon.. we need to finish the project in 10 days, we have just started..
Thanks

AndersBuhl, 2023-04-17:
Hi,

The breathing algorithm is only present in Python and available via Exploration Tool. Also, we have no support for the RZ/V2MA.

nehabhootesh, 2023-04-17:
Hey, may I know where can I get the code for sleep breathing, because I need to add heart rate calculation algorithm for the same file.

nehabhootesh, 2023-04-22:
hey I'm trying to run the python file on command prompt by merging all files, my aim is to jut print the breath rate on the command prompt but have been getting this error.
if data["init_progress"] is not None:
IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
Disconnecting...

Here is the complete code :
import numpy as np
import pg_processor_spoof
from scipy import signal
import acconeer.exptool as et

def main():
    args = et.a111.ExampleArgumentParser().parse_args()
    et.utils.config_logging(args)

    client = et.a111.Client(**et.a111.get_client_args(args))

    client.squeeze = False

    sensor_config = et.a111.IQServiceConfig()
    sensor_config.sensor = args.sensors
    sensor_config.range_interval = [0.2, 1.0]
    sensor_config.profile = sensor_config.Profile.PROFILE_2
    sensor_config.hw_accelerated_average_samples = 20
    sensor_config.downsampling_factor = 2
    sensor_config.update_rate=60
    sensor_config.gain=0.6

    session_info = client.setup_session(sensor_config)
    interrupt_handler = et.utils.ExampleInterruptHandler()
    print("Press Ctrl-C to end session")
    pg_updater = PGUpdater(sensor_config, None, session_info)
    pg_process = pg_processor_spoof.PGProcess(pg_updater)
    pg_process.start()
    client.start_session()
    while not interrupt_handler.got_signal:
        data_info, data = client.get_next()

        try:
            pg_process.put_data(data)
        except pg_processor_spoof.PGProccessDiedException:
            break

    print("Disconnecting...")
    pg_process.close()
    client.disconnect()

class ProcessingConfiguration(et.configbase.ProcessingConfig):
    VERSION = 1

    n_dft = et.configbase.FloatParameter(
        label="Estimation window",
        unit="s",
        default_value=15,
        limits=(2, 20),
        updateable=False,
        order=0,
    )

    t_freq_est = et.configbase.FloatParameter(
        label="Time between estimation",
        unit="s",
        default_value=0.2,
        limits=(0.1, 10),
        updateable=False,
        order=10,
    )

    D = et.configbase.IntParameter(
        label="Distance downsampling",
[... 265 lines cut ...]
        self.config = sensor_config 
   def update(self, data):
        

        if data["init_progress"] is not None:
            self.spect_text_item.setText("Initiating: {} %".format(data["init_progress"]))
        else:
            snr = data["snr"]
            if snr == 0:
                s = "SNR: N/A | {:.0f} dB".format(10 * np.log10(data["lambda_p"]))
            else:
                fmt = "SNR: {:.0f} | {:.0f} dB"
                s = fmt.format(10 * np.log10(snr), 10 * np.log10(data["lambda_p"]))
            print(s)

            f_est = data["f_est"]
            if f_est > 0:
                s = "Latest frequency estimate: {:.2f} Hz | {:.0f} BPM".format(f_est, f_est * 60)
                print(s)

if __name__ == "__main__":
    main()

please help me!! I need to finish the project in 5 days
Thanks !!

nehabhootesh, 2023-04-22:
I copy pasted the same code available in algo but still getting the error 
Press Ctrl-C to end session
Process Process-1:
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1776.0_x64__qbz5n2kfra8p0\lib\multiprocessing\process.py", line 314, in _bootstrap
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1776.0_x64__qbz5n2kfra8p0\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\pg_process.py", line 65, in pg_process_program
    updater.setup(win.ci)
  File "C:\Users\nehab\Desktop\Trial\exacsamecopy.py", line 388, in setup
    self.spect_smax = et.utils.SmoothMax(self.config.update_rate / 15)
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
Disconnecting...

nehabhootesh, 2023-04-22:
Have been getting this error :
PS C:\Users\nehab\Desktop\Trial> python spoof.py -u COM3
Press Ctrl-C to end session
Traceback (most recent call last):
  File "C:\Users\nehab\Desktop\Trial\spoof.py", line 330, in <module>
    out_data = proc(sensor_config, processing_config, session_info, calibration=None).process(data)
  File "C:\Users\nehab\Desktop\Trial\spoof.py", line 102, in __init__
    self.delta_f = 1.0 / n_dft
TypeError: unsupported operand type(s) for /: 'float' and 'FloatParameter'

the complete code:

import numpy as np
import pg_processor_spoof
from scipy import signal
import acconeer.exptool as et

class ProcessingConfiguration(et.configbase.ProcessingConfig):
    VERSION = 1

    n_dft = et.configbase.FloatParameter(
        label="Estimation window",
        unit="s",
        default_value=15,
        limits=(2, 20),
        updateable=False,
        order=0,
    )

    t_freq_est = et.configbase.FloatParameter(
        label="Time between estimation",
        unit="s",
        default_value=0.2,
        limits=(0.1, 10),
        updateable=False,
        order=10,
    )

    D = et.configbase.IntParameter(
        label="Distance downsampling",
        default_value=124,
        limits=(0, 248),
        updateable=False,
        order=20,
    )

    f_high = et.configbase.FloatParameter(
        label="Bandpass high freq",
        unit="Hz",
        default_value=0.8,
        limits=(0, 10),
        updateable=False,
        order=30,
    )

    f_low = et.configbase.FloatParameter(
        label="Bandpass low freq",
        unit="Hz",
        default_value=0.2,
        limits=(0, 10),
        updateable=False,
        order=40,
    )

    lambda_p = et.configbase.FloatParameter(
        label="Threshold: Peak to noise ratio",
        default_value=40,
        limits=(1, 1000),
        updateable=False,
        order=50,
[... 259 lines cut ...]
client.start_session()
while not interrupt_handler.got_signal:
    data_info, data = client.get_next()
    proc=Processor

    try:
        pg_process.put_data(data)
        out_data = proc(sensor_config, processing_config, session_info, calibration=None).process(data)
        data1 = proc(sensor_config, processing_config, session_info, calibration=None).downsample(out_data, n=2)
        P = proc(sensor_config, processing_config, session_info, calibration=None).dft(data1)
        f_est1 = proc(sensor_config, processing_config, session_info, calibration=None).breath_freq_est(P)
        if f_est1 > 0:
            s = "Latest frequency estimate: {:.2f} Hz | {:.0f} BPM".format(f_est1, f_est1 * 60)
            print(s)
    except pg_processor_spoof.PGProccessDiedException:
        break

print("Disconnecting...")
pg_process.close()
client.disconnect()

   
Please resolve our issues, all three different errors. Our goal is to print the breath rate on the command prompt. Please help us in resolving the error.

Thanks!!

nehabhootesh, 2023-04-25:
C:\Users\nehab>python -m acconeer.exptool.a111.algo.sleep_breathing -u COM3
Press Ctrl-C to end session
14:45:08.587 | WARNING | MainProcess      | acconeer.exptool.a111._clients.reg.client | successfully recovered from corrupt frame
14:45:08.660 | WARNING | MainProcess      | acconeer.exptool.a111._clients.reg.client | successfully recovered from corrupt frame
14:45:08.762 | WARNING | MainProcess      | acconeer.exptool.a111._clients.reg.client | successfully recovered from corrupt frame
14:45:09.163 | WARNING | MainProcess      | acconeer.exptool.a111._clients.reg.client | successfully recovered from corrupt frame
14:45:09.237 | WARNING | MainProcess      | acconeer.exptool.a111._clients.reg.client | successfully recovered from corrupt frame
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1776.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1776.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\a111\algo\sleep_breathing\__main__.py", line 9, in <module>
    main(module_info)
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\a111\algo\_standalone_main.py", line 35, in main
    info, sweep = client.get_next()
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\a111\_clients\client.py", line 143, in get_next
    return self.subclient.get_next()
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\a111\_clients\base.py", line 158, in get_next
    return self._get_next()
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\a111\_clients\reg\client.py", line 259, in _get_next
    packet = self._recv_packet(allow_recovery_skip=True)
  File "C:\Users\nehab\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\acconeer\exptool\a111\_clients\reg\client.py", line 402, in _recv_packet
    raise ClientError("got invalid frame and could not recover")
acconeer.exptool.a111._clients.base.ClientError: got invalid frame and could not recover

AndersBuhl, 2023-04-28:
Hi,

I don't see any reason for you to add everything to one script. You should be able to install/transfer the files to the Raspberry PI as is. All modifications done is your responsibility.

nehabhootesh, 2023-05-01:
hey may i know how to run the module on raspberry pi, we have installed acconeer exploration tool on raspberry pi using command. Now we need to run the module!!

AndersBuhl, 2023-05-03:
Hi,

You should be able to use the same steps for running XM112/XB112 on RPi as when running on a Ubuntu laptop.

nehabhootesh, 2023-05-10:
Hey, I have written code to measure heart rate but the accuracy is low. So i need to know is there any calibration to do for the sensor. If yes, how to calibrate let me know the step. If no, how to improve the accuracy.
Thanks regards,
Neha B

closed 2023-06-01 by AndersBuhl