acconeer/acconeer-python-exploration #126: client.get_next() result temperature reading does not change

kangc0016, 2023-04-19:
we run temperature cycle test using XM125 with following python code to collect the result

    client.start_session()
    for i in range(total_samples):                                                  
        result = client.get_next()
    client.stop_session()

however, when i look at the results collected for each sample, the temperature reading does not change, which is incorrect, because the module is put inside of an oven which cycles through a large temperature drift. 

If I use the following code, then the temperature reading from collected result changes, but why should i stop and re-start the session after each sample?
    for i in range(total_samples):  
        client.start_session()                                                
        result = client.get_next()
        client.stop_session()

kangc0016, 2023-04-23:
is this a bug in the python code only or is it also a bug in the c code?

AndersBuhl, 2023-04-24:
Hi,

To know if there is a bug or not I need more information.

Did my suggestion solve your issue?

What is your session configuration?

closed 2023-06-01 by AndersBuhl