acconeer/acconeer-python-exploration #130: XE121 + RaspberryPi as server

pphilbey, 2023-07-06:
Hello, is it possible to run the exploration tool on a machine connected to a RPi w/ XE121?

After combing through the documentation I have the impression the RPi is just not well supported for A121 which is a bit disappointing. Maybe I'm just missing something.

AndersBuhl, 2023-07-07:
Hi,

To be able to connect exploration tool on a Raspberry Pi you should execute the binary 'acc_exploration_server_a121' from the package 'acconeer_rpi_xe121_a121'.
Then you can connect to the RPi using socket connection in exploration tool.

BR,
Anders

pphilbey, 2023-07-10:
Thanks it's working well now, I missed it reading the documentation. The documentation for the exploration tool only mentions A111 for RPi + Server

closed 2023-07-10 by pphilbey