acconeer/acconeer-python-exploration #132: Serial Protocol

31warren, 2023-07-10:
Testing XM125 modules on custom PCBs to break out pins.

Connected via UART.

Flashed the latest .bin file ok to two modules - one module flashed using the tool, another using STM32Cube. (server_a121.bin)

Opened the Exptool and selected Serial but no sub-menu appears to be available to select the Protocol as mentioned in the documentation. Tested separately using Windows and Ubuntu.

The XM125 module is not detected by Exptool (with boot0 disconnected) although it can be flashed when boot0 is connected to +V)

A similar setup for XM132 modules works ok.

Please advise if the Protocol setting is required. Thanks.

Traceback (most recent call last):
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 68, in connect_client
    self.client = a121.Client.open(**open_client_parameters)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\client.py", line 67, in open
    return subclass.open(
           ^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 85, in open
    return cls(client_info=client_info, _override_protocol=_override_protocol)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 115, in __init__
    self._connect_client()
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 141, in _connect_client
    self._server_info = self._retrieve_server_info()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 181, in _retrieve_server_info
    system_info_response = self._send_command_and_wait_for_response(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 294, in _send_command_and_wait_for_response
    return self._wait_for_response(response_type, timeout_s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 312, in _wait_for_response
    for message in self._message_stream:
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 157, in _get_message_stream
    self._link.recv_until(self._protocol.end_sequence)
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\links.py", line 65, in recv_until
    return bytes(super().recv_until(bs=byte_sequence))
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\a111\_clients\links.py", line 272, in recv_until
    raise LinkError("recv timeout")
acconeer.exptool.a111._clients.links.LinkError: recv timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\app\new\backend\_backend.py", line 161, in process_program
    model.execute_task(task)
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 48, in execute_task
    builtin_task(**kwargs)
  File "C:\Users\micha\AppData\Local\Programs\Python\Python311\Lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 80, in connect_client
    raise HandledException(msg)
acconeer.exptool.app.new._exceptions.HandledException: Failed to connect:
recv timeout

vackraetraed, 2023-07-31:
Hello @31warren 

What kind of UART-USB connection do you have?
In Ubuntu, is the COM port detected when connecting it?
After connecting the USB, can you run "sudo dmesg | grep tty" and verify that the COM port is detected for example when I connect XM125 by XE125 I get the following:
[ 8083.645237] usb 1-2: cp210x converter now attached to ttyUSB0
[ 8083.648161] usb 1-2: cp210x converter now attached to ttyUSB1

rastikw, 2023-09-07:
I run into the same issue. XM124 and XM132 work fine, but tested with XM125 and I can flash acc_exploration_server_a121.bin v1.2.0, but Exploration Tool always says:

> Traceback (most recent call last):
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 68, in connect_client
    self.client = a121.Client.open(**open_client_parameters)
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\client.py", line 67, in open
    return subclass.open(
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 85, in open
    return cls(client_info=client_info, _override_protocol=_override_protocol)
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 115, in __init__
    self._connect_client()
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 141, in _connect_client
    self._server_info = self._retrieve_server_info()
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 181, in _retrieve_server_info
    system_info_response = self._send_command_and_wait_for_response(
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 294, in _send_command_and_wait_for_response
    return self._wait_for_response(response_type, timeout_s)
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 312, in _wait_for_response
    for message in self._message_stream:
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\peripherals\communication\exploration_client.py", line 157, in _get_message_stream
    self._link.recv_until(self._protocol.end_sequence)
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\_links\_links.py", line 333, in recv_until
    raise LinkError("recv timeout")
acconeer.exptool._links._links.LinkError: recv timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_backend.py", line 161, in process_program
    model.execute_task(task)
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 48, in execute_task
    builtin_task(**kwargs)
  File "C:\Download\radar\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 80, in connect_client
    raise HandledException(msg)
acconeer.exptool.app.new._exceptions.HandledException: Failed to connect:
recv timeout

skauneboy, 2024-02-01:
It seems like flow control is enabled by default for some reason. You can edit this in serial_link.py:
def __init__(self, port: str, flowcontrol: bool = **True**) -> None:

change to:
def __init__(self, port: str, flowcontrol: bool = **False**) -> None:

Fixed the recv timeout issue I was having.

closed 2024-04-19 by AndersBuhl

ksenthurkumaran18052004, 2024-06-11:
hi , while trying to connect xm12t to accone
[image]
[image]

er exploration tool, its says unable to connect revc timeout. 

but i am able to flash into this.
please tell what to do