acconeer/acconeer-python-exploration #133: GUI bug on linux

HusseinHazem, 2023-08-08:
**Bug**
GUI doesnt not load using linux. The main gui window launches but i get the following `error` when I choose any of the examples on the left bar.

**Traceback**
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/app/new/ui/stream_tab/plugin_widget.py", line 359, in _on_app_model_load_plugin
    self.view_plugin = plugin_spec.create_view_plugin(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 632, in create_view_plugin
    return ViewPlugin(app_model=app_model, view_widget=view_widget)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 367, in __init__
    self.sensor_id_pidget = pidgets.SensorIdPidgetFactory(items=[]).create(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/app/new/ui/plugin_components/pidgets/pidgets.py", line 553, in create
    return SensorIdPidget(self, parent)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/app/new/ui/plugin_components/pidgets/pidgets.py", line 558, in __init__
    super().__init__(factory, parent)
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/app/new/ui/plugin_components/pidgets/pidgets.py", line 522, in __init__
    super().__init__(factory, parent)
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/app/new/ui/plugin_components/pidgets/pidgets.py", line 72, in __init__
    super().__init__(parent=parent)
TypeError: object.__init__() takes exactly one argument (the instance to initialize)
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 511, in on_app_model_update
    self.config_editor.set_data(None)
    ^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'config_editor'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 506, in on_app_model_update
    self.start_button.setEnabled(False)
RuntimeError: Internal C++ object (PySide6.QtWidgets.QPushButton) already deleted.
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 497, in on_backend_state_update
    not_handled = self.config_editor.handle_validation_results(results)
                  ^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'config_editor'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 532, in on_app_model_update
    sensor_config_editor = self.sensor_config_editors[index]
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'sensor_config_editors'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 497, in on_backend_state_update
    not_handled = self.config_editor.handle_validation_results(results)
                  ^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'config_editor'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 532, in on_app_model_update
    sensor_config_editor = self.sensor_config_editors[index]
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'sensor_config_editors'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 532, in on_app_model_update
    sensor_config_editor = self.sensor_config_editors[index]
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'sensor_config_editors'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 497, in on_backend_state_update
    not_handled = self.config_editor.handle_validation_results(results)
                  ^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'config_editor'
Traceback (most recent call last):
  File "/home/hussein/acc-exp-tool-2/.accenv2/lib/python3.11/site-packages/acconeer/exptool/a121/algo/distance/_detector_plugin.py", line 532, in on_app_model_update
    sensor_config_editor = self.sensor_config_editors[index]
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ViewPlugin' object has no attribute 'sensor_config_editors'

**Installed packages**
acconeer-exptool==7.2.1
attrs==23.1.0
beautifulsoup4==4.12.2
bs4==0.0.1
certifi==2023.7.22
charset-normalizer==3.2.0
commonmark==0.9.1
contourpy==1.1.0
cycler==0.11.0
darkdetect==0.7.1
exceptiongroup==1.1.1
fonttools==4.42.0
h5py==3.9.0
idna==3.4
kiwisolver==1.4.4
matplotlib==3.7.2
numpy==1.25.2
packaging==23.1
Pillow==10.0.0
platformdirs==3.10.0
psutil==5.9.5
Pygments==2.16.1
pyparsing==3.0.9
pyperclip==1.8.2
pyqtdarktheme==2.1.0
pyqtgraph==0.13.3
pyserial==3.5
PySide6==6.5.2
PySide6-Addons==6.5.2
PySide6-Essentials==6.5.2
python-dateutil==2.8.2
pyusb==1.2.1
PyYAML==6.0.1
QtAwesome==1.2.3
QtPy==2.3.1
requests==2.31.0
rich==12.6.0
scipy==1.11.1
shiboken6==6.5.2
six==1.16.0
soupsieve==2.4.1
textual==0.1.18
typing_extensions==4.7.1
urllib3==2.0.4

**Environment:**
 - Host OS: arch linux
 - Acconeer hardware: xm125
 - Acconeer software version: a121-v1.1.0
 - Exploration version: v7.2.1

AndersBuhl, 2023-08-08:
Hi @HusseinHazem 

This seems to be an issue with PySide6>=6.5. So please try to install version 6.4.3.

BR,
Anders

AndersBuhl, 2023-08-09:
Hi,

From the error message is seems the device is not connected.

Can you provide some screen dumps when this happens?
What version of RSS and Exploration Tool are you running?
Which HW are you using?
How was the HW flashed?

HusseinHazem, 2023-08-18:
Hi Anders, sorry for the late reply. I am already added to group dialout and i already flashed the  firmware with acc_exploratation_server_a121.bin via the gui tool from the flash tab. I also am "connected" in the stream tab so the device seems connected. im using ET 7.2.1 and rss: a121-v1.1.0

[image]
[image]

AndersBuhl, 2023-08-18:
Hi,

So the exceptions appear when you press "Start measurement"?
Can you try to update Exploration Tool to v7.3.0?

BR,
Anders

closed 2023-08-18 by AndersBuhl