acconeer/acconeer-python-exploration #135: Exploration tool crashes

sebglatz, 2023-12-15:
**Describe the bug**
`python -m acconeer.exptool.app` opens the GUI.
But when clicking the A121 window the GUI crashes.
(Would not happen when clicking the A111 window)

**Traceback**
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/__main__.py", line 11, in <module>
    from .new import main
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/__init__.py", line 13, in <module>
    from .app import main
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/app.py", line 24, in <module>
    from .ui import AppModelViewer, MainWindow
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/__init__.py", line 4, in <module>
    from . import components, icons, utils
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/components/__init__.py", line 4, in <module>
    from . import a121, pidgets
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/components/a121/__init__.py", line 7, in <module>
    from .sensor_config_editor import SensorConfigEditor
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/components/a121/sensor_config_editor.py", line 19, in <module>
    from acconeer.exptool.app.new.ui.components.json_save_load_buttons import (
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/components/json_save_load_buttons/__init__.py", line 4, in <module>
    from .buttons import (
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/components/json_save_load_buttons/buttons.py", line 19, in <module>
    from acconeer.exptool.app.new.ui.icons import CHECKMARK, FOLDER_OPEN, SAVE
  File "/home/seb/.local/lib/python3.10/site-packages/acconeer/exptool/app/new/ui/icons.py", line 6, in <module>
    import qtawesome as qta
  File "/usr/local/lib/python3.10/dist-packages/qtawesome/__init__.py", line 19, in <module>
    from .iconic_font import IconicFont, set_global_defaults
  File "/usr/local/lib/python3.10/dist-packages/qtawesome/iconic_font.py", line 22, in <module>
    from qtpy.QtCore import QObject, QPoint, QRect, qRound, Qt
ImportError: cannot import name 'qRound' from 'qtpy.QtCore' (/home/seb/.local/lib/python3.10/site-packages/qtpy/QtCore.py)

**Installed packages**
acconeer-exptool @ file:///home/seb/acconeer-python-exploration
action-msgs==1.2.1
action-tutorials-interfaces==0.20.3
action-tutorials-py==0.20.3
actionlib-msgs==4.2.3
airium==0.2.5
ament-cmake-test==1.3.5
ament-copyright==0.12.8
ament-cppcheck==0.12.8
ament-cpplint==0.12.8
ament-flake8==0.12.8
ament-index-python==1.4.0
ament-lint==0.12.8
ament-lint-cmake==0.12.8
ament-package==0.14.0
ament-pep257==0.12.8
ament-uncrustify==0.12.8
ament-xmllint==0.12.8
angles==1.15.0
aniso8601==9.0.1
ansible-base==2.10.8
anyio==3.6.2
apache-libcloud==3.2.0
appdirs==1.4.4
apturl==0.5.2
aqtinstall==3.1.6
argcomplete==1.8.1
arghelper==0.4.2
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arm-gdb==0.9.9
arrow==1.2.3
[... 518 lines cut ...]
visualization-msgs==4.2.3
wadllib==1.3.6
wcwidth==0.2.5
webcolors==1.13
webencodings==0.5.1
websocket-client==1.5.1
Werkzeug==2.3.7
wirerope==0.4.7
wrapt==1.13.3
WTForms==3.0.1
wxPython==4.2.0
xdg==5
xkit==0.0.0
xlwt==1.3.0
xmltodict==0.12.0
yamllint==1.26.3
yoctopuce==1.10.55677
zeroconf==0.39.0
zipp==1.0.0
zstandard==0.21.0

**Environment (please complete the following information):**
 - Host OS: Ubuntu 22.04.3 LTS
 - Acconeer hardware: XE125 (But GUI crashes independently of connected HW)
 - Exploration version: v7.7.3

vackraetraed, 2023-12-18:
Most Exploration Tool errors related to GUI is usually related to a new version of PySide6, I would recommend trying:
python -m pip install PySide6==6.4.3

Best regards,
Anton

sebglatz, 2023-12-26:
This pyside6 version caused another crash on my system.
But a clean install inside virtualenv with python version 3.11.5 solves the problem (pyside 6.6.1 and exptool 7.7.3)

closed 2023-12-26 by sebglatz