acconeer/acconeer-python-exploration #138: Sparkfun X125 board does not function in Exploration Tool

cewick70, 2024-04-05:
Sparkfun X125 Module does not run any available Stream function under Exploration tool

What has happened: Exploration Tool was placed on both a Windows 11 machine and a NVIDIA Orin machine to test the a Sparkfun X125 Module.  Both OS and machines produced similar results:

Exploration Tool appears to flash the module ok when the "flash" option is selected.  The most current Exploration Server binary file was flashed to the X125 Module per directions and without incident and the module connected ok via serial link, but then under "Stream" when any function of Exploration Tool is selected  (e.g. Distance measurement) an error window appears.
For example, when calibrate sensor  is pushed with distance detector selected a "calibration failed " window appears with the following traceback: 

Traceback (most recent call last):
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\algo\distance\_detector_plugin.py", line 211, in calibrate_detector
    self._detector_instance.calibrate_detector()
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\algo\distance\_detector.py", line 541, in calibrate_detector
    self._calibrate_offset()
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\algo\distance\_detector.py", line 767, in _calibrate_offset
    self.client.setup_session(session_config)
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\communication\exploration_client.py", line 208, in setup_session
    setup_response = self._server_stream.wait_for_message(a121_messages.SetupResponse)
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\_core\communication\message_stream.py", line 64, in wait_for_message
    for message in self._stream:
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\_core\communication\message_stream.py", line 100, in _get_stream
    resp = self.protocol.parse_message(header, payload)
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\_core\communication\exploration_protocol\_latest.py", line 59, in parse_message
    raise RuntimeError(f"Could not parse response with header:\n{header}")
RuntimeError: Could not parse response with header:
{'status': 'ok', 'tick_period': 0, 'payload_size': 0, 'calibration_info': [{'ensor_id': 1, 'temperature': 24, 'data': '9f9d09000400030007002300250024002300010025002a002e002b0026000200050006000200390a4b085d085d07050b380cdb08c307b1ff11049a00e7040702eafb2cfa37f5bff3dafa8bf3d3f152f534f1aefacff84610f28900007100000032353700a60000004f00000044151b0089000000de1f35000e0000009cea160068000100050000003b000000172ab701000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'}], 'metadata': [[{'calibration_temperature': 24, 'frame_data_length': 50, 'sweep_data_length': 50, 'max_sweep_rate': 168.674713134766, 'high_speed_mode': False, 'base_step_length_m': 0.00250227400101721, 'subsweep_data_offset': [0], 'subsweep_data_length': [50]}]]}

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_backend.py", line 161, in process_program
    model.execute_task(task)
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 56, in execute_task
    plugin_task(**kwargs)
  File "C:\XM125\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\algo\distance\_detector_plugin.py", line 213, in calibrate_detector
    raise HandledException("Failed to calibrate detector") from exc
acconeer.exptool.app.new._exceptions.HandledException: Failed to calibrate detector
g traceback:

When start measurement is pushed an error window appears with a "cannot start" message.
All other Exploration Tool functions fail in a similar fashion.

vackraetraed, 2024-04-05:
Hi @cewick70 

The header looks kind of corrupted.
In the Exploration Tool you can limit the baudrate of the serial device.
It can be accessed by clicking the cogwheel in the middle of the port selection and "Connect"-button.
Try limiting the baudrate to 115200 and see what happens.

cewick70, 2024-04-05:
That was it!  Distance measurement seems to be working now at 115200.
Autobaud must have been selecting something much too high for
board/machine.  Thank you!
CEW

vackraetraed, 2024-04-08:
Glad that helped!

The XM125 and Exploration Tool kicks up the baudrate to 2000000.
From looking at the design, it looks like the component has 2000000 as max baudrate, perhaps things get weird when trying to push the component to its limits.

You can try increasing the baudrate until it fails if you want to find some optimal value if you need higher throughput than 115200.

closed 2024-04-19 by AndersBuhl