acconeer/acconeer-python-exploration #139: Options for running on M2 MacOS

saillingaway, 2024-04-13:
Hey there, I'd really like to use acconeer for a project but I'm on an M2 machine. From the readme it seems like there isn't out of the box support for any MacOS, silicon or not, and I'm not sure what the best way to get it up and running would be. 

I'm assuming I could set up a virtual environment/machine with an acconeer-compatible linux/ubuntu os and just follow the install settings, but I don't know how the virtual environment will impact acconeer's performance or what other tradeoffs there may be. Is there a specific set up you recommend? Apologies if the answer is somewhere else already, I'm somewhat new to this kind of development and appreciate your patience and help.

saillingaway, 2024-04-15:
Hey @vackraetraed, thanks for the quick response, that's awesome.

I did experience that issue you mentioned with `python -m pip install --upgrade acconeer-exptool[app]`, zsh complains about it: **zsh: no matches found: acconeer-exptool[app]**. When it failed I ended up just trying the generic `pip install acconeer-exptool` from the [pypi](https://pypi.org/project/acconeer-exptool/) page which _did_ work but I bailed a bit through on the install so I never finished it.

I think the special escape character may not have made it into your correction, but escaping the `[` like you said worked for me with a simple backslash: `python3 -m pip install --upgrade acconeer-exptool\[app]`.

I'll try the other things you mentioned and once I've got it working I'll share what I did so other Mac users can enjoy the tool as well :)

saillingaway, 2024-04-15:
> ...but as long as you do not use XC120, it should be fairly straight forward to resolve issues you might face.

Could you actually expand a little on this? I'm planning to use the XE125 EVK which I believe should be A121 though, so I should be safe there I hope.

vackraetraed, 2024-04-16:
Hello @saillingaway 

Yes sorry, I seem to have missed the"\" in my correction. Running the command "python3 -m pip install --upgrade acconeer-exptool\[app]" as you did should be the correct command.

Yes, please let us know if you get it running and if you face any issues which were not solved by my previous suggestions.

XE125 uses serial COM port over the USB, I got that working fairly easily with no changes to Python code. Only the additional installation steps.
The XC120 (XE121 has the radar and is placed on XC120) however uses pure USB rather than a serial COM port. Which complicates things, I could not get it running within the day I tried running different things on Mac.

So I would suggest sticking to XE125 since it should work without major complications.

saillingaway, 2024-05-05:
@vackraetraed Coming back with a question, not sure if you can help. The linux setup involves these three pieces:

> Setup up permissions needed for UART communication by adding the current user to the 'dialout' group.
> $ sudo usermod -a -G dialout [USERNAME]
>
> Create an udev rule for SPI communication with XM112.
> Will make sure that /etc/udev/rules.d/50-ft4222.rules contains this content:
> SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="601c", MODE:="0666"
>
> Create an udev rule for USB communication with XC120.
> Will make sure that /etc/udev/rules.d/50-xc120.rules contains this content:
>
>  SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a41d", MODE:="0666"
>  SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a42c", MODE:="0666"
>  SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a42d", MODE:="0666"
>  SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a449", MODE:="0666"

Changing permissions on the device file that represents the serial port so that the relevant user can read/write seems straightforward, but I'm less sure how to handle the second and third things here. As far as I've been able to tell there isn't something like udev rules for MacOS. Do you remember how you handled this?

vackraetraed, 2024-05-06:
Hello @saillingaway 

I did not really do any of those.
But the first one with setting UART access, was handled through some GUI interface on Mac.
Unfortunately I did not pay much attention here since the system just kind of asked me to fill in super user password to give the access. It essentially just guided me to some kind of device manager when I tried to access the UART and told me to fill in the password to give it access.
Make sure that the VCOM port drivers are installed before trying to access it.

As for the rest with the udev rules, it is only for XC120 (USB) and XM112 (SPI), but I do not think you will be using either of those hardwares so you should be able to skip that. I did not have time during the day I borrowed a Mac to get XC120 working.

saillingaway, 2024-05-06:
@vackraetraed I see now that leaving out the dev related steps make sense since they're for the XC120 and XM112 like you said, should have caught that. 

Just want to double check something. The VCOM drivers I have installed are the [silicon labs drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads) for Mac, which I believe should work for the USB-UART COM ports on the XE-125. Those are compatible, correct? 

I'm able to start the Exploration tool up with `python3 -m acconeer.exptool.app` and select the A121 option but it hangs with the following error:

```
Traceback (most recent call last):
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/app_model/port_updater.py", line 36, in timerEvent
    usb_devices = get_usb_devices()  # type: ignore[name-defined]
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_core/communication/comm_devices.py", line 201, in get_usb_devices
    for device_vid, device_pid, serial_number in pyusbcomm.iterate_devices():
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_pyusb/pyusbcomm.py", line 23, in iterate_devices
    dev = usb.core.find(find_all=True)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/usb/core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

This seems like an indication that pyusb can't find a library on my system to interface with USB devices. However, I have libusb version 1.0.27 installed (installed with homebrew), but it could be a permissions problem?

saillingaway, 2024-05-06:
Also just want to confirm, were either of the machines that you got it working on silicon macs?

saillingaway, 2024-05-06:
Fixed the **[no backend available error from above](https://github.com/acconeer/acconeer-python-exploration/issues/139#issuecomment-2096837350)**, I was on the right track. [libusb doesn't know to look where Homebrew puts libusb files on install](https://stackoverflow.com/questions/70729330/python-on-m1-mbp-trying-to-connect-to-usb-devices-nobackenderror-no-backend-a):

> Installing libusb with Homebrew is not sufficient. Homebrew puts the relevant files in /opt/homebrew/Cellar/libusb/1.0.24/lib and creates symbolic links in /opt/homebrew/lib. But pyusb is not aware of these paths.
> 
> You have two main options:
> 
> Add /opt/homebrew/lib to the environment variable DYLD_LIBRARY_PATH. For a permanent setup, add it to ~/.zshenv:
> `export DYLD_LIBRARY_PATH="/opt/homebrew/lib:$DYLD_LIBRARY_PATH"`
> Create a symbolic link in your home directory. This takes advantage of the fact that ~/lib is a default fallback path for libraries:
> `ln -s /opt/homebrew/lib ~/lib`

I opted for the second and it solved that problem, however now when I try to connect in the tool, it throws this error:

```
Traceback (most recent call last):
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/backend/_model.py", line 75, in connect_client
    self.client = client_factory()
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/a121/_core/communication/client.py", line 50, in open
    return super().open(
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_core/communication/client.py", line 87, in open
    return subclass.open(  # type: ignore[no-any-return, attr-defined]
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 91, in open
    return cls(client_info=client_info)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 115, in __init__
    self._server_info = self._retrieve_server_info()
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 139, in _retrieve_server_info
    system_info_response = self._server_stream.wait_for_message(messages.SystemInfoResponse)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 64, in wait_for_message
    for message in self._stream:
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 83, in _get_stream
    self._error_callback(e)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 135, in _close_before_reraise
    raise exception
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 81, in _get_stream
    header_in_bytes = self._link.recv_until(self.protocol.end_sequence)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/_core/communication/links/serial_link.py", line 155, in recv_until
    raise LinkError("recv timeout")
acconeer.exptool._core.communication.links.buffered_link.LinkError: recv timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/backend/_backend.py", line 161, in process_program
    model.execute_task(task)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/backend/_model.py", line 51, in execute_task
    builtin_task(**kwargs)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/backend/_model.py", line 87, in connect_client
    raise HandledException(msg)
acconeer.exptool.app.new._exceptions.HandledException: Failed to connect:
recv timeout

```

I found issue #132 where the recv timeout was solved by [changing one line in serial_link.py](https://github.com/acconeer/acconeer-python-exploration/issues/132#issuecomment-1921364205), but I can't locate where the acconeer files on my machine actually are, so I can't change them. Here's a picture of the selections I had if it helps:

[image]

I also tried to grab the most recent binary to flash and see if it would work but that also threw an error:

```
Traceback (most recent call last):
File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/ui/flash_tab/threads.py", line 56, in run
flash_image(
File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/flash/_flasher.py", line 92, in flash_image
PRODUCT_NAME_TO_FLASH_MAP[serial_device_name].flash(
File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/flash/_stm32uart/_stm32flasher.py", line 52, in flash
flasher.sync()
File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/flash/_stm32uart/_stm32flasher.py", line 162, in sync
raise Stm32FlashException(
acconeer.exptool.flash._stm32uart._stm32flasher.Stm32FlashException: DFU synchronization failed for '/dev/cu.SLAB_USBtoUART', make sure device is in DFU mode
```
 
I'm pretty sure it was in DFU mode, I followed the instructions that popped up. If you have any ideas on what to try please let me know.

saillingaway, 2024-05-07:
> I found issue #132 where the recv timeout was solved by [changing one line in serial_link.py](https://github.com/acconeer/acconeer-python-exploration/issues/132#issuecomment-1921364205), but I can't locate where the acconeer files on my machine actually are, so I can't change them.

I found where this file was and tried their fix, unfortunately didn't work for me.

vackraetraed, 2024-05-08:
Hello @saillingaway 

You have pretty much covered the steps I took to run it now.
The correct drivers seem to be installed and the backend libusb library seems to have been installed as well.
So at this point we might have to get creative.

Have you flashed an Exploration Server binary on XM125 before?
If you have, Exploration Tool only works with 1 of the UARTs so I suggest trying both when trying to connect.
Also try limiting the baudrate to 115200 to begin with by using the cogwheel next to the connect button.

The issue you are trying to reference seem connected to users trying to use the Exploration Server/Tool without flow control (i.e. not using CTS/RTS pins, since they did not use the Evaluation kit and did not connect these pins, all of our modules use flow control to achieve higher baudrates)

vackraetraed, 2024-05-08:
@saillingaway 

You can try downloading the Exploration Server binary for XM125 from our developer site then run the command:
`python -m acconeer.exptool.flash flash -d XM125 -i acc_exploration_server_a121.bin`

Before running this command, you can hold down the DFU button while connecting the USB. This will trigger the bootloader, but not use the RESET button (in case for some reason the RESET button is broken).
Then you can try running the command.

saillingaway, 2024-05-08:
@vackraetraed Thanks for the reply.

> Have you flashed an Exploration Server binary on XM125 before?
> If you have, Exploration Tool only works with 1 of the UARTs so I suggest trying both when trying to connect.

I haven't flashed anything on XM125, this is the first time I'm working with anything from Acconeer in general. I did try both connection types but threw the same double recv timeout error. The options I saw were UART and UART3, although sometimes it's UART and UART4 (pictured in the previous reply).

Just confirming, when you say there's no pre-flashed binary that means the hardware didn't come flashed with anything? I'll try downloading the binary, connecting it with the just the DFU button held down and see what happens with that command.

Also I want to rule out this problem being a permissions issue. Could the `recv timeout` error be indicating that my user **doesn't** have necessary permissions to access the device file? In other words, does the Exploration tool throw a specific error for insufficient permissions, or would it just timeout...? I tried to confirm that my user has the permissions to access the file by using `screen /dev/tty.SLAB_USBtoUART3` and looking at the access permissions, owner and user, I _should_ have permission but I'm just a little suspicious still. If you have any ideas how to test this let me know.

saillingaway, 2024-05-09:
> You can try downloading the Exploration Server binary for XM125 from our developer site then run the command:
`python -m acconeer.exptool.flash flash -d XM125 -i acc_exploration_server_a121.bin`

I tried downloading the binary and then running the command (ACCONEER XM125 EXPLORATION SERVER A121 from [this page](https://developer.acconeer.com/home/a121-docs-software/xm125-xe125/)) but unfortunately it seems like there's some trouble detecting the device. I tried adding what I think was the port, but it didn't make a difference, same error:

[image]

mtntrx, 2024-05-12:
I'd like to bring the exploration tool up as well on an M1 Mac.  I'll post my progress, where I am stuck, and see if it helps. I'm considering a switch to this IC/platform for a current product, but will need help.

HW: 

- M1 mac (air)
- SparkFun XM125 dev board, with CH340C USB-to-serial converter.  They imply this should be compatible with the Exploration Tool (https://www.sparkfun.com/products/24540)
- USB-C connection

SW: SMTCubeProgrammer: UART, cu.usbserial-10, 1115200, E81 off RTS/DTR Low
FW: **successfully flashd i2c_distance_detector.bin**

**Basic functionality success:**

Results: I am able to get distance measurements read out via I2C(SDA,SCL) and an external host controller (ESP8266)
Distance behavior is mixed, so I'd like to get the explorer up and running.  

**Notes:**

It seems I have all I need to commuincate with the device (show by successful connection and flash)
That said, there are dev board diffs that the exploration tool will expose?
e.g. CP2105-F01-GM Silcon labs chip vs. the CH340C (https://developer.acconeer.com/download/xe125-product-brief/)

**Exploration tool attempt**
- following saillingaway's lead
- I've only done marginally extra things vs sailing away, but thought the DFU confirmation and alternative connect success might be useful

Note: FW on deivce is still i2c_distance_detector.bin
```
git clone git@github.com:acconeer/acconeer-python-exploration.git
## v7.10.0, acconeer-python-exploration
cd acconeer-python-exploration
python3 virtual env setup `ptyhon3 -m venv venv`
python3 -m venv venv
. venv/bin/activate
python3 -m pip install --upgrade pip
python3 -m pip install --upgrade acconeer-exptool\[app]
python3 -m acconeer.exptool.app
```

In the app:

- select A121
- select my serial port: cu.usbserial-10
- settings->baudrate: 1115200
- connect

Result: same rcv timeout error as seen by saillingaway

edited: acconeer-python-exploration/src/acconeer/exptool/_core/communication/links/serial_link.py line 102
  def __init__(self, port: str, flowcontrol: bool = True) -> None:
  to
  def __init__(self, port: str, flowcontrol: bool = False) -> None:

Result: no change; same rcv timeout error as seen by saillingaway

I'm able to confirm DFU mode by

1. Connecting via SMTCubeProgrammer (at 115200): SUCCESS
2. Disonnect/close SMTCubeProgrammer
3. Connecting via acconeer.exptool.app (at 115200): FAIL
4. Connecting via SMTCubeProgrammer (at 115200): SUCCESS

So, not much more progress than sailingaway,  just the "ability to connect" and DFU confirmation.  I'll keep trying from here.  One thing to confirm:  The device should connect from the exptool independent of what FW is loaded?

vackraetraed, 2024-05-14:
Hi @mtntrx 

>     1. When the python Acconeer Exploration Tool (GUI) connects to the device, it is expecting the device to be OUT of DFU mode, correct? i.e. it's not in flash mode, rather it's in application mode?
> 
>     2. The acc_exploration_server_a121.bin needs to be flashed for the device to communicate to the Acconeer Exploration Tool (GUI)?  This seems likely, but I don't recall seeing that specifically called out in the GUI
> 
>     3. The acc_exploration_server_a121.bin FW will try to commuincate at 115200, with hardware flow control on for starters and "negotiate" up?
> 
>     4. Given the step `acc_exploration_server_a121.bin`, there is a reasonable chance we have different versions of some libraries?  How long since you have had success here?  I'm banging away at this primarily due to the fact you said you had success. (even though not formally supported)
> 
>     5. Is there any chance you could find time to start a fresh git-pull, venv, upgrade, and check functionality?  That might help rule out version diffs as an issue.

1. Yes, it is usually the safest to reset the device after flashing.
2. Yes, in order to use the Exploration Tool, you need the Exploration Server to be flashed on XM125.
3. Yes, it starts on 115200 with flow control then tries to kick it up to 2000000 with flow control. The sparkfun board uses a UART to USB converter that does not handle 2000000 very well, so limiting the baud rate when using the sparkfun board is a good idea.
4. I tried it around February this year.
5. I don't actually have a Mac available, I just tried it for fun with a borrowed Mac. I could possibly borrow a friends M1 or M2 (not sure which one he has) and try to see if I can get it working.

tyVandeZande, 2024-05-16:
Hi y'all, Im working with @saillingaway to try to get this working on Apple Silicon. I'm good at finding hacks, but bad at solid solutions. I'm on Sonoma 14.3.1 on an M1 w/ a factory-fkdup HDMI issue

mtntrx, 2024-05-18:
@tyVandeZande welcome to the fun.  I made it a big further then @saillingaway (standing/wobbling on their shoulders) but I'm still stuck.  I have flashing working in 2 tools, so I know I have a working serial port via the /dev device.  That said, the connection being used for communicating with the exploration firmware just times out.  If you have any other ideas of things to try I'll try anything.  I'm out of ideas at the moment.

I busted out a windows machine and got instantly stuck there with something stupid on my end I'm sure.  I just filed an issue for it -- will probably be embarrassed when I figure out what is wrong there.

tyVandeZande, 2024-05-18:
@mtntrx Hi, I was able to get it working last night. I had my list of steps, but it got deleted before I posted. Working again on it tonight. Can You please share what type of mac are you using it on? Macbook pro vs macbook air vs imac, etc?

tyVandeZande, 2024-05-19:
@saillingaway @mtntrx 
Is this the error you're receiving?

`Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/app/new/backend/_model.py", line 75, in connect_client
    self.client = client_factory()
                  ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/client.py", line 50, in open
    return super().open(
           ^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/_core/communication/client.py", line 87, in open
    return subclass.open(  # type: ignore[no-any-return, attr-defined]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 91, in open
    return cls(client_info=client_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 115, in __init__
    self._server_info = self._retrieve_server_info()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 139, in _retrieve_server_info
    system_info_response = self._server_stream.wait_for_message(messages.SystemInfoResponse)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 64, in wait_for_message
    for message in self._stream:
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 83, in _get_stream
    self._error_callback(e)
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 135, in _close_before_reraise
    raise exception
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 81, in _get_stream
    header_in_bytes = self._link.recv_until(self.protocol.end_sequence)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/_core/communication/links/serial_link.py", line 155, in recv_until
    raise LinkError("recv timeout")
acconeer.exptool._core.communication.links.buffered_link.LinkError: recv timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/app/new/backend/_backend.py", line 161, in process_program
    model.execute_task(task)
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/app/new/backend/_model.py", line 51, in execute_task
    builtin_task(**kwargs)
  File "/opt/homebrew/lib/python3.12/site-packages/acconeer/exptool/app/new/backend/_model.py", line 87, in connect_client
    raise HandledException(msg)
acconeer.exptool.app.new._exceptions.HandledException: Failed to connect:
recv timeout
`

tyVandeZande, 2024-05-19:
One step I remember having to do to that was not great was using `--break-system-packages` to install the evaluation tool.
I don't think that's relevant to you, if you already have the GUI tool up and running. 

The timeout you're receiving is a symptom @saillingaway had when the firmware was not flashed. 

You've confirmed the switch on the board is set to `USB` ?
- Try flicking the USB/v-tech switch once or twice & or make sure you've tried the USB.C cable in both orientations & or make sure the cable is secure/making good contact. 
- I've had the device recognized, connect, and stream if I plug it in before or after the tool runs. 

Component missing on hardware? (I will post a picture of my board)

saillingaway, 2024-05-19:
@mtntrx @vackraetraed wanted to give an update on progress @tyVandeZande and I made this morning. Here's exactly what we did (in order) and the results. Mostly recording all of this information so I can come back to refer to it later if need be, but someone else may find it helpful, who knows.

---
We used a (kind of low quality) usbc-to-usb data cable and an [Apple usb-to-usbc adapter](https://www.apple.com/shop/product/MJ1M2AM/A/usb-c-to-usb-adapter) to connect the eval kit to the computer. We also are only using the **upper left usbc port on the machine** next to the charging port because we've had inconsistencies/problems with connecting to/detecting other devices (mainly esp32s) on both of our machines using any other port, like @tyVandeZande mentioned. The kit was already flashed with the binary mentioned above using STM32CubeProgrammer, so thank you for telling us about that.

1. Plugged the adaptor into the machine. In the Exploration Tool, selected Serial, /dev/tty.SLAB_USBtoUART, clicked **Connect**. Resulted in the same recv timeout error @tyVandeZande and I posted above ([x](https://github.com/acconeer/acconeer-python-exploration/issues/139#issuecomment-2097077038), [x](https://github.com/acconeer/acconeer-python-exploration/issues/139#issuecomment-2119068218)).
2. Unplugged and replugged cable from eval kit. Selected Serial, /dev/tty.SLAB_USBtoUART, clicked **Connect**. Resulted in same recv timeout error.
3. Unplugged and replugged adaptor from machine. Selected Serial, /dev/tty.SLAB_USBtoUART, clicked **Connect**. Resulted in same recv timeout error.
4. Unplugged adaptor from machine, flipped it over and plugged it back in. Selected Serial, /dev/tty.SLAB_USBtoUART, clicked **Connect**. **It connected and we were able to select a module, start measurements and view live feedback from the kit.**

**Machine:** 2023 M2 Pro Macbook
**macOS:** Ventura 13.4

We think your issue could be the cable. When we picked it up the eval kit at one point the Exploration Tool threw an error that was something like **Error: Failed to get next frame**.

dthakur, 2024-05-20:
Just want to +1 that the behavior in my case is identical to @mtntrx.
1. Device showing as `cu.usbserial-10`.
2. I can flash using the exploration tool in DFU mode (did not try stm32cube).
3. I am using SparkFun XM125 dev board.
4. I get `recv timeout` whether I limit the baud rate using the cog or not.
I'm on M2 + Sonoma using `libusb 1.0.27` and `python 3.11`.

mtntrx, 2024-05-25:
Hey folks, finally success for me.

I'm going to say my MAC/M1 issue was some combo of driver and the USBtoUART chip (CH340C on the Spark Fun module)

I was able to get the module to work successfully on a Windows machine, but not to full BAUD rate.  (success upward of 460,800) and that seem to be good enough.

So, it felt like a MAC driver issue.  I did not try any other drivers for that CH340C. (maybe that would have been easier)

Instead I took the harder road.  with some tricky soldering I wired in a CP2102 USBtoUART chip.  I was able to flash, but still failed to connect to the explorer tool.  The difference in device names was telling (me: cu.usbserial-0001 @saillingaway:  /dev/tty.SLAB_USABtoUART)

So, I downloaded/installed the SiLab drivers and I'm up with 
* either usb slot
* usbc->a adapter, A->C cable (don't ask, it's just what was on my bench top)
* stand alone XM125 module, no dev board
* [CP2102 USB-serial adapter](https://www.amazon.com/dp/B0C8RRDCXB?psc=1&smid=AXSDPEI0SD5BE&ref_=chk_typ_imgToDp)  wired in to the 4 UART pins
* 468000 baud (auto did not work, but my hacked in connections are not great, so that could be the limiter)
* SiLabs driver, selecting /dev/tty.SLAB_USABtoUART

Software install steps:
```
git clone git@github.com:acconeer/acconeer-python-exploration.git
## v7.10.0, acconeer-python-exploration
cd acconeer-python-exploration
python3 virtual env setup `ptyhon3 -m venv venv`
python3 -m venv venv
. venv/bin/activate
python3 -m pip install --upgrade pip
python3 -m pip install --upgrade acconeer-exptool\[app]
python3 -m acconeer.exptool.app
```

Thanks all for the info.  Knowing success was possible helped a lot.

saillingaway, 2024-05-25:
@mtntrx nice! Glad it's working for you. @dthakur do you have the SiLabs driver? Seems like that fixed things for them.

dthakur, 2024-05-25:
I'm glad things worked out for @mtntrx, but if I understand correctly, they switched the USB to serial chip from CH340C to CP2102 (from Silicon Labs). I don't intend to perform that hardware modification, so I don't think SiLab drivers are relevant for me. Perhaps I'm mistaken?

mtntrx, 2024-05-25:
@dthakur I think you are correct, a SiLabs driver won't help.
But, since we know that sparkFun dev board USBtoUART chip CAN work (as demo'ed by the windows success) there is a chance an improved driver for the CH340C could help.  I just didn't try one ... yet.

dthakur, 2024-05-26:
Solved for me.

Solution: Install 340C drivers from [here](https://www.wch-ic.com//downloads/CH34XSER_MAC_ZIP.html).

This pops up a new device, which works just fine.
[image]

Thanks @mtntrx and @saillingaway.

saillingaway, 2024-05-26:
@dthakur oops, re-reading and yes I misunderstood what they wrote. Happy to hear you got it working!