acconeer/acconeer-python-exploration #140: Cannot install acconeer-python-exploration on Raspbian OS (Raspberry pi)

yalatoom, 2024-04-18:
I tried to install the acconeer-python-exploration on Raspbian OS on Raspberry pi 4 using:

```
python3 -m pip install --upgrade acconeer-exptool[app]
```

but I got this error :
```
Could not find a version that satisfies the requirement pyside6!=6.3.0,!=6.3.2,<6.5; extra == "app" (from acconeer-exptool[app]) (from versions: )
No matching distribution found for pyside6!=6.3.0,!=6.3.2,<6.5; extra == "app" (from acconeer-exptool[app])
```

My raspberry pi info:
```
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: 37ba4b21ba2c4cc0bc62a8f7686ff7a7
           Boot ID: cd0860ef5cd1454db4605cf389eb3319
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.17-v7l+
```

AndersBuhl, 2024-04-19:
Hi,

There is no support for the application to run on Raspberry PI, although it is possible to run algo-lib with scripts.

The you need to install the algo-part only.
`python3 -m pip install --upgrade acconeer-exptool[app]`

BR,
Anders

yalatoom, 2024-04-22:
You mean this:
`
python3 -m pip install --upgrade acconeer-exptool[algo]
`
but after importing the ibrary i got this error:

`
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/__init__.py", line 4, in <module>
    from . import a111, utils
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a111/__init__.py", line 6, in <module>
    from . import recording
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a111/recording.py", line 15, in <module>
    import h5py
  File "/home/pi/.local/lib/python3.7/site-packages/h5py/__init__.py", line 25, in <module>
    from . import _errors
ImportError: libhdf5_serial.so.103: cannot open shared object file: No such file or directory
`

so I did the following to solve it:

```
sudo apt-get install libhdf5-dev

pip3 uninstall h5py

pip3 install h5py
```
it is installed now

closed 2024-04-22 by yalatoom