acconeer/acconeer-python-exploration #142: Trying to calculate distance but get "AttributeError: module 'acconeer.exptool.a121' has no attribute 'DetectorConfig'

yalatoom, 2024-04-22:
I tried to use this code based on your example to calculate distance:

```
import acconeer.exptool as et
from acconeer.exptool import a121
from acconeer.exptool.a121.algo.distance import Detector, DetectorConfig, ThresholdMethod

SENSOR_ID = 1

def main():
    # Specify the connection method and device manually
    serial_port = "/dev/ttyUSB1"  # Example: specify the serial port for USB connection

    # Open the connection to the sensor
    client = a121.Client.open(serial_port=serial_port)

    # Other parts of the script remain the same
    detector_config = a121.DetectorConfig(
        start_m=0.0,
        end_m=2.0,
        max_profile=a121.Profile.PROFILE_3,
        max_step_length=12,
        threshold_method=a121.ThresholdMethod.RECORDED,
    )
    detector = a121.Detector(client=client, sensor_ids=[SENSOR_ID], detector_config=detector_config)

    detector.calibrate_detector()

    detector.start()

    interrupt_handler = et.utils.ExampleInterruptHandler()
    print("Press Ctrl-C to end session")

    while not interrupt_handler.got_signal:
        detector_result = detector.get_next()
        try:
            if detector_result is not None:
                for distance in detector_result[SENSOR_ID].distances:
                    print("Distance:", distance)
        except et.PGProccessDiedException:
            break

    detector.stop()

    print("Disconnecting...")
    client.close()

if __name__ == "__main__":
    main()
```
but get this error:

`AttributeError: module 'acconeer.exptool.a121' has no attribute 'DetectorConfig'`

I am using Raspberry Pi and connected "SparkFun Qwiic Pulsed Coherent Radar Sensor - Acconeer XM125" with USB port to my Raspberry Pi.

AndersBuhl, 2024-04-23:
Line 15 should say:
`detector_config = DetectorConfig(`
and  not 
`detector_config = a121.DetectorConfig(`

yalatoom, 2024-04-23:
> Line 15 should say: `detector_config = DetectorConfig(` and not `detector_config = a121.DetectorConfig(`

Thank you. It solved that error but received another error:

```
Traceback (most recent call last):
  File "/home/pi/Desktop/A121.py", line 46, in <module>
    main()
  File "/home/pi/Desktop/A121.py", line 12, in main
    client = a121.Client.open(serial_port=serial_port)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/_core/communication/client.py", line 93, in open
    override_baudrate,
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 87, in open
    return cls(client_info=client_info)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 111, in __init__
    self._server_info = self._retrieve_server_info()
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 135, in _retrieve_server_info
    system_info_response = self._server_stream.wait_for_message(messages.SystemInfoResponse)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 64, in wait_for_message
    for message in self._stream:
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 85, in _get_stream
    self._error_callback(e)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 131, in _close_before_reraise
    raise exception
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 82, in _get_stream
    self._link.recv_until(self.protocol.end_sequence)
  File "/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/_core/communication/links/serial_link.py", line 154, in recv_until
    raise LinkError("recv timeout")
acconeer.exptool._core.communication.links.buffered_link.LinkError: recv timeout
```

AndersBuhl, 2024-04-24:
Try lowering the baudrate to get a more stable connection:

```
    # Open the connection to the sensor
    client = a121.Client.open(serial_port=serial_port, override_baudrate=115200)
```

yalatoom, 2024-04-25:
> Try lowering the baudrate to get a more stable connection:
> 
> ```
>     # Open the connection to the sensor
>     client = a121.Client.open(serial_port=serial_port, override_baudrate=115200)
> ```

Thank you. That solve my issue and it works now.

closed 2024-04-25 by yalatoom