acconeer/acconeer-python-exploration #143: Vibration Example Error - ImportError: cannot import name 'get_high_frequency_sensor_config' from 'acconeer.exptool.a121.algo.vibration'

yalatoom, 2024-04-25:
I tried to run this code based on Vibration Example but Import Error showed:
```
# Copyright (c) Acconeer AB, 2022-2023
# All rights reserved

from __future__ import annotations

import numpy as np

# Added here to force pyqtgraph to choose PySide
#import PySide6  # noqa: F401

#import pyqtgraph as pg

import acconeer.exptool as et
from acconeer.exptool import a121
from acconeer.exptool.a121.algo._utils import APPROX_BASE_STEP_LENGTH_M
from acconeer.exptool.a121.algo.vibration import (
    Processor,
    ProcessorConfig,
    ProcessorContext,
    ProcessorResult,
    get_high_frequency_sensor_config,
)

class PrintUpdater:
    def setup(self, win):
        pass

    def update(self, processor_result: ProcessorResult) -> None:
        # Extra result
        time_series = processor_result.extra_result.zm_time_series
        lp_displacements_threshold = processor_result.extra_result.lp_displacements_threshold
        amplitude_threshold = processor_result.extra_result.amplitude_threshold

        # Processor result
        lp_displacements = processor_result.lp_displacements
        max_amplitude = processor_result.max_sweep_amplitude
        max_displacement = processor_result.max_displacement
        max_displacement_freq = processor_result.max_displacement_freq
        time_series_rms = processor_result.time_series_std

        # Print object presence metric
        print("Max amplitude:", max_amplitude)
        print("Amplitude threshold:", amplitude_threshold)

        # Print time series
        if time_series is not None and amplitude_threshold < max_amplitude:
            assert time_series_rms is not None
            print("RMS:", time_series_rms)
            print("Time series:", time_series)

        # Print spectrum
        if lp_displacements is not None:
            assert time_series is not None
            assert lp_displacements is not None

            print("Low pass displacements:", lp_displacements)
            print("Low pass displacements freqs:", processor_result.lp_displacements_freqs)
            print("Low pass displacements threshold:", lp_displacements_threshold)

            if max_displacement_freq is not None and max_displacement is not None:
                print("Max displacement frequency:", max_displacement_freq)
                print("Max displacement:", max_displacement)

def main():
    # Specify the connection method and device manually
    serial_port = "/dev/ttyUSB1"  # Example: specify the serial port for USB connection

    # Open the connection to the sensor
    client = a121.Client.open(serial_port=serial_port, override_baudrate=115200)

    args = a121.ExampleArgumentParser().parse_args()
    et.utils.config_logging(args)

    sensor_config = get_high_frequency_sensor_config()

    metadata = client.setup_session(sensor_config)

    processor = Processor(
        sensor_config=sensor_config,
        metadata=metadata,
        processor_config=ProcessorConfig(),
        context=ProcessorContext(),
    )
    updater = PrintUpdater()

    client.start_session()
    interrupt_handler = et.utils.ExampleInterruptHandler()
    print("Press Ctrl-C to end session")

    while not interrupt_handler.got_signal:
        result = client.get_next()
        plot_data = processor.process(result)
        try:
            updater.update(plot_data)
        except et.PGProccessDiedException:
            break

    print("Disconnecting...")
    client.close()

if __name__ == "__main__":
    main()
```

Error:
`ImportError: cannot import name 'get_high_frequency_sensor_config' from 'acconeer.exptool.a121.algo.vibration' (/home/pi/.local/lib/python3.7/site-packages/acconeer/exptool/a121/algo/vibration/__init__.py)`

Using Sparkfun XM125 with Raspberry Pi 4 Model B connected through USB.

I tried to check if the module " existed but it is not:

```
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import acconeer.exptool.a121.algo.vibration as vibration_algo
>>> print(dir(vibration_algo))
['Processor', 'ProcessorConfig', 'ProcessorContext', 'ProcessorResult', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '_load_algo_data', '_processor', 'get_sensor_config']
>>> 

```

AndersBuhl, 2024-04-26:
Hi,

I cannot reproduce this.  For me, this is available.
What version of exploration tool are you running?

Also, I see that you are running python 3.7, which we don't fully support anymore.

BR,
Anders

yalatoom, 2024-05-02:
I am using acconeer.exptool version 7.6.1
I solved the problem by adjusting the configuration manually like this:
```
# Get processor configuration
    processor_config = ProcessorConfig(
        time_series_length=1024,
        lp_coeff=0.95,
        sensitivity=100.0,
        amplitude_threshold=1.0
    )
```
but I have another question which is about the unit of "time_series" in the example that you provided here "https://github.com/acconeer/acconeer-python-exploration/blob/master/examples/a121/algo/vibration/vibration.py"
is it in mm or micro meter?

AndersBuhl, 2024-05-06:
the unit is number of samples. so the in this case the time series will be a buffer with 1024 samples

yalatoom, 2024-05-06:
> the unit is number of samples. so the in this case the time series will be a buffer with 1024 samples

No, it is displacement value not number of samples. My question what is the unit of displacement in "time_series" variable that been used in the vibration example.

AndersBuhl, 2024-05-06:
the time between samples in the time series will be 1 / frame_rate, if that is what you are asking for

yalatoom, 2024-05-06:
> the time between samples in the time series will be 1 / frame_rate, if that is what you are asking for

No not that. I am not asking about time. The sensor can measure object displacement as been done in the vibration example. I want to know what is the unit of displacement there. Is it in mm or micro meter? That is my question.

AndersBuhl, 2024-05-06:
The displacement is not an output from the algorithm. You only get the estimated frequency

AndersBuhl, 2024-05-06:
The unit for the displacement is µm

closed 2024-05-23 by AndersBuhl