acconeer/acconeer-python-exploration #145: Debugging custom plugins & launching exploration tool from cmd line with multiple plugins with --plugin-module my_plugin

saillingaway, 2024-05-22:
I realize this may not be the best place to ask these general questions, but didn't have any luck looking on the docs or in the `my_plugin.py` example itself in `examples/app/new/plugins`. I'm really liking how flexible and comprehensive the exploration tool is so far, so I'm hoping someone can give me a little help on this. 

I currently launch the tool from terminal with the command `python3 -m acconeer.exptool.app.new` and using `python3 -m acconeer.exptool.app.new --plugin-module my_plugin_name` from the [add my own plugin guide](https://docs.acconeer.com/en/latest/exploration_tool/algo/a121/adding_your_own_plugin/index.html) and it works as expected, the plugin appears in the tool when it's launched. Is there CLI support for launching the exploration tool with multiple plugins? Or is there a way to just reload the tool? 

I'd like to know what the most intuitive and easy way would be to be able to test and switch between multiple plugins with only one eval kit. If there isn't cli support for this I figure I can just set up a bash script that will take some arguments and start whatever plugins specified, but (I think) those would all be separate exploration tool instances.

I'm also not sure what the options are in terms of debugging my custom plugins. Any information on that would be really appreciated. Thanks in advance, and again just really liking the tool so far in general.

andrehedesand, 2024-05-23:
Hi there @saillingaway !

We're glad you're enjoying Exploration Tool, so are we :)

So there are a couple of way to go about loading multiple plugins:
1. *Repeat the `--plugin-module` option:* This will load all the plugins you specify, e.g. `python3 -m acconeer.exptool.app.new --plugin-module A --plugin-module B` will load both python modules `A` and `B`. This isn't clear from the guide or the `--help`. We'll make sure to clarify that till next release!
2. *Bundle together plugins:* What Exploration Tool does when given a `--plugin-module` is
import that Python module and call the `register()` function within... **That's it!** You can register multiple plugins in the `register()` function (try duplicating the line `register_plugin(MY_PLUGIN)`!)

Exploration Tool isn't easily debugged, I'm afraid. We usually end up relying on the old trusty `print`-debugging or using the *app model viewer* (`--amv`) to identify GUI desyncs.

saillingaway, 2024-05-23:
@andrehedesand thanks for the quick reply!

Sounds pretty straightforward, I'll give it a try later.

saillingaway, 2024-05-23:
Running into an issue trying to bundle two plugins. I have both plugin python files inside `acconeer/examples/app/new/plugins/` and in **plugin_A.py** I'm registering both plugins like this:

```
def register() -> None:
    register_plugin(PLUGIN_A)
    register_plugin(PLUGIN_B)
```

plugin_B doesn't have any register function. When I run `python3 -m acconeer.exptool.app.new --plugin-module plugin_A` I get the error **NameError: name 'PLUGIN_B' is not defined**. The spelling of my plugin is the same everywhere. Is there somewhere else I need to import **plugin_B**, or does **plugin_B** need it's own register empty register function even though it'll be registered when **plugin_A** is being set up? Thanks in advance.

Here's the full traceback:

```
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/__main__.py", line 8, in <module>
    main()
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/app.py", line 53, in main
    import_and_register_plugin_module(plugin_module_name)
  File "/Users/sophie/Library/Python/3.9/lib/python/site-packages/acconeer/exptool/app/new/plugin_loader.py", line 54, in import_and_register_plugin_module
    imported_module.register()
  File "/Users/sophie/acconeer-python-exploration-master/examples/app/new/plugins/phase_tracking_plugin.py", line 285, in register
    register_plugin(DISTANCE_DETECTOR_PLUGIN_DOTBLISS)
NameError: name 'DISTANCE_DETECTOR_PLUGIN_DOTBLISS' is not defined
```

andrehedesand, 2024-05-24:
Hi again!

Thank you for your clear description!

Make sure that the Python-file that contains the `register()` function (`plugin_A.py`, right?) also imports `PLUGIN_B`.

We wholly rely on Python's import system here, which also means that you can structure your plugins in any way you would structure Python code.

I'll give a few examples below (with a lot of things emitted):

# 1. Your situation (from what I can gather):

```py
# plugin_B.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_B = PluginSpec(...)
```
```py
# plugin_A.py

from plugin_B import PLUGIN_B  # maybe this import is missing?

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_A = PluginSpec(...)

def register():
    register_plugin(PLUGIN_A)
    register_plugin(PLUGIN_B)
```

Here you should be able to run `python3 -m acconeer.exptool.app.new --plugin-module plugin_A` to get both plugins.

# 2. Separated plugins
```py
# plugin_B.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_B = PluginSpec(...)

def register():
    register_plugin(PLUGIN_B)
```
```py
# plugin_A.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_A = PluginSpec(...)

def register():
    register_plugin(PLUGIN_A)
```

Here you should be able to run `python3 -m acconeer.exptool.app.new --plugin-module plugin_A --plugin-module plugin_B` to get both plugins.

# 3. Separate file that bundles
```py
# plugin_B.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_B = PluginSpec(...)
```
```py
# plugin_A.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_A = PluginSpec(...)
```
```py
# bundle.py

from plugin_A import PLUGIN_A
from plugin_B import PLUGIN_B

def register():
    register_plugin(PLUGIN_A)
    register_plugin(PLUGIN_B)
```
Here you should be able to run `python3 -m acconeer.exptool.app.new --plugin-module bundle` to get both plugins.

# 4. Hybrid of **2.** and **3.**
```py
# plugin_B.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_B = PluginSpec(...)

def register():
    register_plugin(PLUGIN_B)
```
```py
# plugin_A.py

class BackendPlugin(ProcessorBackendPluginBase): ...
class ViewPlugin(ProcessorViewPluginBase): ...
class PlotPlugin(PgPlotPlugin): ...

class PluginSpec(PluginSpecBase): ...

PLUGIN_A = PluginSpec(...)

def register():
    register_plugin(PLUGIN_A)
```
```py
# bundle.py

from plugin_A import register as register_A
from plugin_B import register as register_B

def register():
    register_A()
    register_B()
```

Here you should be able to run `python3 -m acconeer.exptool.app.new --plugin-module bundle` or `python3 -m acconeer.exptool.app.new --plugin-module plugin_A --plugin-module plugin_B` to get both plugins.

I hope this helps :)