acconeer/acconeer-python-exploration #146: i am not able to connect to xm125, it says unable to connect recv timeout

ksenthurkumaran18052004, 2024-06-11:
i am not able to connect to xm125, it says unable to connect recv timeout
[image]
But i amble to flash
please help

vackraetraed, 2024-06-13:
Hello @ksenthurkumaran18052004 

Are you using XE125 or have you used XM125 directly and try to make your own connecting board?

sulaco67, 2024-06-15:
Hi. Exact same issue here. Using XE125 board with Silabs 11.3.0.176 driver. Please assist

trent-lane, 2024-07-24:
Hi to all! I've encountered the same problem. The module is barebone XM125 with 3v3 power and UART levels, CH340 is used for serial connection.
[image]

Connections: 
```
1    VIN    3v3
2    GND    GND
3    1v8    1v8
4    GND    GND
5    TX     RX_CH340
6    RX     TX_CH430
13   GND    GND
14   nRST   button to GND
26   BOOT0  button to 3v3
``` 
I am able to flash it: 
[image]

But when trying to connect all I get is "recv timeuot". I've tried to set baudrates manually, but 9600, 57600 or 115200 didn't work. The oscilloscope shows that TX pin of CH340 goes low for a brief time (XM125's RX), but nothing changes at it's RX (XM125's TX, accordingly).

vackraetraed, 2024-07-24:
@sulaco67 There are two COM ports on XE125, try using both. Both are able to flash the module but only the enhanced COM port supports running the Exploration Tool.

@trent-lane In a similar manner, only one of the UARTs support the Exploration Tool protocol, which UART TX and RX have you connected? Could it be the one that only supports flashing?
Both the default module binary and the Exploration Tool uses flow control, you can either connect the the flow control pins or disable it by re-building the Exploration Server binary without flow control in the XM125 SDK.

trent-lane, 2024-07-24:
> @trent-lane In a similar manner, only one of the UARTs support the Exploration Tool protocol, which UART TX and RX have you connected?

First I've tried:

```
5    UART_TX     RX_CH340
6    UART_RX     TX_CH430
``` 
My next try was:
```
21    DEBUG_UART_RX     TX_CH340
23    DEBUG_UART_TX     RX_CH430
``` 
Pin mapping and names are taken from "Entry+ Module XM125 Datasheet, v1.2".

I can flash the module using either UART, but still can't connect it even when using DEBUG_UART connection. I've also tried some common baudrates and swapped to another USB ports.

[image]

trent-lane, 2024-07-24:
I've found a solution. There are additional pins that need to be connected for bootloader to communicate with desktop app:
```
8    UART_CTS     DTR_CH340
9    UART_RTS     CTS_CH430
``` 
Thank you, [vackraetraed](https://github.com/vackraetraed)!

vackraetraed, 2024-07-24:
@trent-lane Glad it helped with flow control.

As for the other peoples issue, I am not sure it helps.
XE125 has flow control automatically setup on the USB.
However, if they are wiring UART from the header pins, they should use CTS/RTS.
Also, solder bridges on XE125 have to be cut for the UART pins if the header pins are used.
Otherwise the on-board UART->USB chip will be driving the pins. 
Looking at the schematic, it is specifically SB1, SB2, SB3, SB4.

caspersauce, 2024-08-09:
@vackraetraed @sulaco67 Was this resolved?

I am getting the exact same issue on Windows 10. I have a brand new XE125, a fresh install of the explorer tool, fresh install of the CP210X drivers. The tool opens just fine, flashes the most recent firmware just fine, but refuses to connect. I get "recv timeout" no matter what I do. Both ports give the same error, despite baud rate, flow control etc...

caspersauce, 2024-08-12:
No idea if this helps anyone else, but I reflashed both ports multiple times, still no progress. Unplugged everything, rebooted everything, no progress.

I then flipped the USB-C over because of something I saw in another post. Everything now works... No idea what this means, but maybe it will help somebody.

skypopcorn1, 2024-12-05:
flipping the USB-C over fixed the issue for me as well. Hours wasted trying to debug this. @acconneer team - any thoughts on this?

vackraetraed, 2024-12-06:
@skypopcorn1 
It is not unthinkable that parts of the connector might be broken which could potentially break the symmetry of the USB-C connection.

Glad to hear that it still works though.

Jean-BaptistePerrier, 2024-12-12:
Fixed by swapping the usb cable as well. I'm betting the DP/DM pins are only routed for one row of the connector.

vackraetraed, 2024-12-13:
@caspersauce @skypopcorn1 @Jean-BaptistePerrier 

If possible, could you scan the QR codes and post the serial numbers of your XE125 devices?

Also, any info on the USB cables used would help as well.

closed 2025-04-10 by AndersBuhl

alexandrebebon, 2025-05-19:
Hi,
Same problem, and same solution. 
Doesn't work (and works again when flipping cable) with various cables (no-brands ones, Bose headphone one...). Seems not to come from the cable.
SN of the XE125 device (or at least numbers that I could find on the packaging and on the board): 

1891-XE125-ND (on the packaging)
XE125 PB41 v1.3 (on the board)
XE125R1DNL20242300253 (QRcode on the board)

Best regards

vackraetraed, 2025-05-19:
Hi @alexandrebebon 

We would be very interested in receiving one of these devices to look closer at the issue.
If possible we would like to swap your device for a new XE125.

If you would be willing to help us with this, please contact: info@acconeer.com in order to sort out details.

Best regards,
Anton