acconeer/acconeer-python-exploration #147: Sparse IQ measurement start failure

erian747, 2024-07-04:
**Describe the bug**
Trying to evaluate A121 sensor by using XM125 ev kit but have some problems running the Exploration tool, see second screenshot below
After running a measurement (which itself ends in a warning "failed to get_next", see first screenshot) i can not start a new Sparse IQ measurement without restarting the Exploration tool
Distance measurement which i want to evaluate i am not able to run even after a fresh start

Traceback (most recent call last):
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_backend.py", line 161, in process_program
    model.execute_task(task)
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\backend\_model.py", line 56, in execute_task
    plugin_task(**kwargs)
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\a121\algo\_plugins\_a121.py", line 163, in start_session
    raise RuntimeError("Client is not attached. Can not 'start'.")
RuntimeError: Client is not attached. Can not 'start'.

**To reproduce**
Steps to reproduce the behavior:
1. Start application by running run_app.bat from cmd
2. Go to 'Stream'
3. Click on 'Sparse IQ', successful some times after UI restart
4. Run service again
5. See error

**Traceback**
If applicable, add any exception tracebacks.
Traceback (most recent call last):
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 383, in _handle_backend_message
    self._handle_backend_general_message(message)
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 402, in _handle_backend_general_message
    self._update_saveable_file(message.data)
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 426, in _update_saveable_file
    self.saveable_file.unlink()
  File "pathlib.py", line 1354, in unlink
PermissionError: [WinError 32] Det går inte att komma åt filen eftersom den
används av en annan process: 'C:\\Users\\erian\\AppData\\Local\\Acconeer AB\\acconeer_exptool\\plugoneer\\temp\\582046d1-3234-41d9-aa9d-e5abebd7608b.h5'
Traceback (most recent call last):
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 279, in stop
    self.load_plugin(None)
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 742, in load_plugin
    self._unload_current_plugin()
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 731, in _unload_current_plugin
    self._update_saveable_file(None)
  File "C:\jobb\acconeer\tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\app_model\app_model.py", line 426, in _update_saveable_file
    self.saveable_file.unlink()
  File "pathlib.py", line 1354, in unlink
PermissionError: [WinError 32] Det går inte att komma åt filen eftersom den
används av en annan process: 'C:\\Users\\erian\\AppData\\Local\\Acconeer AB\\acconeer_exptool\\plugoneer\\temp\\582046d1-3234-41d9-aa9d-e5abebd7608b.h5'
QThread: Destroyed while thread is still running
Press any key to continue . . .
**Installed packages**
If applicable, add a list of installed packages. Obtain the list with `pip freeze`.
pyserial==3.5

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**
 - Host OS: Windows 10
 - Acconeer hardware: XM125
 - Acconeer software version: [e.g. 1.7.0]
 - Exploration version: Latest

**Additional context**
Add any other context about the problem here.

williserdman, 2024-07-09:
Hey, not at all a developer but had this same issue and was just able to get around it. Found the answer from another issue that was erroneously closed https://github.com/acconeer/acconeer-python-exploration/issues/138.

tl;dr: try setting the baud rate to 115200

erian747, 2024-07-09:
Thanks, with 115200 it works fine

vackraetraed, 2024-07-10:
Hi @erian747 

The issue which @williserdman mentions uses a Sparkfun board which has a faulty hardware integration.
My suggestion to reduce baudrate was because that board does not support default baudrate, but it does not address the original issue that Sparkfun should re-design their board for full functionality.

In your case, you only mention "XM125".
You shortly mention "XM125 ev kit", does this refer to the XE125? Are you actually using XE125?
Either case, setting 115200 baudrate is a work-around that does not address the core issue.
Because either there is a sub-optimal integration that does not support 2000000 baudrate.
Or if XE125 is used, something might be wrong with it. Or perhaps the cable is too long.

But if you are OK with only running at 115200, then I suppose we do not have to dig deeper.

erian747, 2024-07-10:
In my case measurement speed requirement is low so 115200 will do fine too

williserdman, 2024-07-10:
@vackraetraed is it worth it/possible to add a note to Acconeer's documentation of SparkFun board's issue? #138, myself and this thread have all had issues because of this. I only ask as even though it's not a failure of the XM125, the evaluation tool crashes and makes it feel as if it is. SparkFun documentation for this board is essentially "here's a link to Acconeer docs"

closed 2025-04-10 by AndersBuhl