acconeer/acconeer-python-exploration #149: Support for machine learning interface on new chip A121

HuyHung0, 2024-07-19:
Hello,

I am currently working on a project that involves deep learning and real-time data acquisition using the A121 radar sensor (the XE125 Evaluation Board). I noticed that your previous branch (v3) supported an interface for machine learning with the A111 radar sensor. It would be highly beneficial if similar support could be provided for the A121 radar sensor.

Additionally, I would appreciate it if you could share the reasons behind the discontinuation of the machine learning interface support. Understanding this would be incredibly helpful.

AndersBuhl, 2024-07-25:
Hi,

The machine learning interface did not serve the purpose we wanted. Instead we recommend checking out the standalone machine learning examples:

[Gesture control](https://github.com/acconeer/acconeer-a121-gesture-control)
[Grass detection](https://github.com/acconeer/acconeer-a121-grass-detection)
[Carpet detection](https://github.com/acconeer/acconeer-a121-carpet-detection)

Hope it helps!

HuyHung0, 2024-07-26:
Thank you. I have checked those repo before.

closed 2024-07-26 by HuyHung0