acconeer/acconeer-python-exploration #151: Unable to use cached data with breathing algorithm

fhudson96, 2024-07-24:
In order to try and develop off of the breathing rate algorithm, I would like to be able to have some cached sensor results to then load from and test algorithms on. I'm wondering the best approach to do this? Currently I am:
- Running through with a recoder to get values 
- Loading in those recorded values, mocking the client and then replacing the client get_next() with the cached data 

With this setup the presence detector never outputs true - whereas it does when I just run the device as usual. And hence no breathing information is ever outputted either.

Code below is what I am trying to use.

Thanks!

`USE_RECORDING=False
RUN_RECORDING=False
def main():
    args = a121.ExampleArgumentParser().parse_args()
    et.utils.config_logging(args)

    # Setup the configurations
    # Detailed at https://docs.acconeer.com/en/latest/exploration_tool/algo/a121/ref_apps/breathing.html

    # Sensor selections
    sensor = 1

    # Ref App Configurations
    breathing_processor_config = BreathingProcessorConfig(
        lowest_breathing_rate=6,
        highest_breathing_rate=60,
        time_series_length_s=20,
    )

    # Presence Configurations
    presence_config = PresenceProcessorConfig(
        intra_detection_threshold=4,
        intra_frame_time_const=0.15,
        inter_frame_fast_cutoff=20,
        inter_frame_slow_cutoff=0.2,
        inter_frame_deviation_time_const=0.5,
    )

    # Breathing Configurations
    ref_app_config = RefAppConfig(
        use_presence_processor=True,
        num_distances_to_analyze=3,
        distance_determination_duration=5,
        breathing_config=breathing_processor_config,
        presence_config=presence_config,
    )

    # End setup configurations

    # Preparation for client
    sensor_config = get_sensor_config(ref_app_config=ref_app_config)
    filename = "data_breathing_tester_2.h5"
    record = None
    if USE_RECORDING:
        client = a121.Client.open(mock=True)
        record = a121.load_record(filename)
    else:
        client = a121.Client.open(**a121.get_client_args(args))
    client.setup_session(sensor_config)

    # Preparation for reference application processor
    ref_app = RefApp(client=client, sensor_id=sensor, ref_app_config=ref_app_config)
    recorder = a121.H5Recorder(filename) if RUN_RECORDING else None
    ref_app.start(recorder=recorder)

    interrupt_handler = et.utils.ExampleInterruptHandler()
    print("Press Ctrl-C to end session")

    while not interrupt_handler.got_signal:
        if record is None:
            result = client.get_next()
        else:
            result = record.results.__next__()
        assert not isinstance(result, list)

        processor_result = ref_app.processor.process(result)

        processed_data = RefAppResult(
            app_state=processor_result.app_state,
            distances_being_analyzed=processor_result.distances_being_analyzed,
            breathing_result=processor_result.breathing_result,
            presence_result=processor_result.presence_result,
        )
        try:
            print("Presence result " + str(processed_data.presence_result.presence_detected))
            print("Breathing result " + str(processed_data.breathing_result))
        except et.PGProccessDiedException:
            break

    ref_app.stop()
    print("Disconnecting...")
    client.close()

if __name__ == "__main__":
    main()`

AndersBuhl, 2024-07-25:
Hi,

The best way to replay a recorded file is to use the [_ReplayingClient](https://github.com/acconeer/acconeer-python-exploration/blob/master/src/acconeer/exptool/a121/_core_ext/_replaying_client.py#L29). You could look at [post_process_distance_result.py](https://github.com/acconeer/acconeer-python-exploration/blob/master/examples/a121/algo/distance/post_process_distance_result.py) to get a hint on how to use it.

BR,
Anders

fhudson96, 2024-07-25:
Hi Anders,

Thanks for your response, this was really helpful!

I will close this issue and just pop the working code below if anyone else needs it.

Cheers!
Finlay

```
# Copyright (c) Acconeer AB, 2022-2023
# All rights reserved

from __future__ import annotations
import h5py

import acconeer.exptool as et
from acconeer.exptool import a121
from acconeer.exptool.a121.algo.breathing import RefApp
from acconeer.exptool.a121.algo.breathing._ref_app import (
    BreathingProcessorConfig,
    RefAppConfig,
    get_sensor_config, RefAppResult, _record_algo_data,
)
from acconeer.exptool.a121 import H5Record, _ReplayingClient, _StopReplay
from acconeer.exptool.a121.algo.presence import ProcessorConfig as PresenceProcessorConfig

USE_RECORDING = True
RUN_RECORDING = False

def runner(client, ref_app):
    interrupt_handler = et.utils.ExampleInterruptHandler()
    print("Press Ctrl-C to end session")
    while not interrupt_handler.got_signal:
        try:
            result = client.get_next()
            assert not isinstance(result, list)

            processor_result = ref_app.processor.process(result)

            processed_data = RefAppResult(
                app_state=processor_result.app_state,
                distances_being_analyzed=processor_result.distances_being_analyzed,
                breathing_result=processor_result.breathing_result,
                presence_result=processor_result.presence_result,
            )
            print("Presence result " + str(processed_data.presence_result.presence_detected))
            # print("Breathing result " + str(processed_data.breathing_result))
            if processed_data.breathing_result is not None:
                print(processed_data.breathing_result.breathing_rate)
        except (et.PGProccessDiedException, _StopReplay):
            break

def get_ref_app_from_record(file):
    record = H5Record(file)
    algo_group = file["algo"]
    sensor_ids = algo_group["ref_app_sensor_id"][()].tolist()
    ref_app_config = RefAppConfig.from_json(algo_group["ref_app_config"][()])
    client = _ReplayingClient(record, realtime_replay=False)
    ref_app = RefApp(client=client, sensor_id=sensor_ids, ref_app_config=ref_app_config)
    return client, ref_app, ref_app_config

def main():
    args = a121.ExampleArgumentParser().parse_args()
    et.utils.config_logging(args)

    # Setup the configurations
    # Detailed at https://docs.acconeer.com/en/latest/exploration_tool/algo/a121/ref_apps/breathing.html

    # Preparation for client
    filename = "data_breathing_tester_2.h5"
    if USE_RECORDING:
        # client = a121.Client.open(mock=True)
        with h5py.File(filename, "r") as file:
            # TODO PUT ALL THIS IN FUNCTIONS
            client, ref_app, ref_app_config = get_ref_app_from_record(file)
            ref_app.start()
            runner(client, ref_app)
        # TODO TO WORK IT ALL NEEDS TO BE IN THIS WITH STATEMENT WHICH IS NOT IDEAL
    else:
        client = a121.Client.open(**a121.get_client_args(args))

        # Sensor selections
        sensor = 1

        # Ref App Configurations
        breathing_processor_config = BreathingProcessorConfig(
            lowest_breathing_rate=6,
            highest_breathing_rate=60,
            time_series_length_s=20,
        )

        # Presence Configurations
        presence_config = PresenceProcessorConfig(
            intra_detection_threshold=4,
            intra_frame_time_const=0.15,
            inter_frame_fast_cutoff=20,
            inter_frame_slow_cutoff=0.2,
            inter_frame_deviation_time_const=0.5,
        )

        # Breathing Configurations
        ref_app_config = RefAppConfig(
            use_presence_processor=True,
            num_distances_to_analyze=3,
            distance_determination_duration=5,
            breathing_config=breathing_processor_config,
            presence_config=presence_config,
        )

        # End setup configurations
        sensor_config = get_sensor_config(ref_app_config=ref_app_config)
        client.setup_session(sensor_config)

        # Preparation for reference application processor
        ref_app = RefApp(client=client, sensor_id=sensor, ref_app_config=ref_app_config)
        recorder = a121.H5Recorder(filename) if RUN_RECORDING else None
        if recorder is not None:
            recorder.require_algo_group("breathing")
        ref_app.start(recorder=recorder)

        runner(client, ref_app)

        if RUN_RECORDING:
            # Manually add algo group after recording is complete
            with h5py.File(filename, 'a') as file:
                if 'algo' not in file:
                    print("Creating algo group")
                    algo_group = file.create_group('algo')
                else:
                    algo_group = file['algo']
                    print("Algo group already exists")

                print("Write algo group to h5 file structure")
                _record_algo_data(algo_group, sensor, ref_app_config)

    ref_app.stop()
    print("Disconnecting...")
    client.close()

if __name__ == "__main__":
    main()
````