acconeer/acconeer-python-exploration #152: Can not flash XM125 DFU not working

Test-BMOHB, 2024-08-03:
I've purchased a XM125 EV  kit from [mouser.com](http://mouser.com/) and I've attached the XM125 IOT module and have been trying to flash it with the explorer tool but it will not flash.  I have installed all of the silicon labs drivers on my windows machine, it recognizes the com port and I have made sure my USB connection supports data.  I have followed all the instructions, followed the sequence of DFU and reset buttons, and have reinstalled python, the explorer tool, and restarted my laptop several times but with no success.  Please assist as I don't see anything in the FAQ to remedy this issue.  Also, what part numbers are for the ribbons for the xe121, I thought cables came with it but it doesn't and I would like to connect it and evaluate it.
Error MSG:

Traceback (most recent call last):
  File "C:\Users\ticke\Downloads\portable_exploration_tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\app\new\ui\flash_tab\threads.py", line 56, in run
    flash_image(
  File "C:\Users\ticke\Downloads\portable_exploration_tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\flash\_flasher.py", line 92, in flash_image
    PRODUCT_NAME_TO_FLASH_MAP[serial_device_name].flash(
  File "C:\Users\ticke\Downloads\portable_exploration_tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\flash\_stm32uart\_stm32flasher.py", line 52, in flash
    flasher.sync()
  File "C:\Users\ticke\Downloads\portable_exploration_tool\tools\python-3.9.10-embed-amd64\lib\site-packages\acconeer\exptool\flash\_stm32uart\_stm32flasher.py", line 162, in sync
    raise Stm32FlashException(
acconeer.exptool.flash._stm32uart._stm32flasher.Stm32FlashException: DFU synchronization failed for 'COM11', make sure device is in DFU mode

[image]

vackraetraed, 2024-08-12:
Hello @Test-BMOHB 

It looks similar to as if the bootloader is not triggered. If you are unlucky, one of the buttons RESET or DFU might be broken.
You can try to work around the RESET button by holding down the DFU button followed by connecting the USB instead of pressing the RESET button. Or switching from external power source to USB power source while holding down the DFU button.

As for the XE121, XE121 is meant to be used with an MCU evaluation board like a NUCLEO which has Arduino UNO headers. It can also be used with XC120 or a Raspberry PI.
If extra sensors beyond the on-board sensor is needed, then XS121 can be bought. This comes with a flex cable which can be attached to the XE121.

Best regards,
Anton

Test-BMOHB, 2024-08-13:
Hello @vackraetraed ,
Thanks for the response.  I tried both suggestions and it doesn't work.  I'll request an exchange from the retailer I purchased it from.

Test-BMOHB, 2024-08-13:
Closed as the hardware defective.

closed 2024-08-13 by Test-BMOHB