acconeer/acconeer-python-exploration #153: how to determine the needed distance from object for vibration measurement using A121(Python)

yalatoom, 2024-08-06:
How to determine the needed distance from an object for vibration measurement using A121 (python)?
what maximum distance the A121 sensor can estimate the vibration of an object?
I found that in the below URL that variable [measured_point] could handle that, but what is the unit of it?

(https://docs.acconeer.com/en/latest/example_apps/a121/vibration.html#acconeer.exptool.a121.algo.vibration._example_app.ExampleAppConfig.measured_point)

AndersBuhl, 2024-08-12:
Hi,

measured_point is unit less since it refers to the measured radar point. The points has 2.5 mm between each other so measured_point=80 corresponds to 0.2 meters. The maximum distance will depend on the displacement of the object and also what type of object it is. Good reflectors will have a longer maximum distance.

BR,
Anders

yalatoom, 2024-08-12:
> Hi,
>
> measured_point is unit less since it refers to the measured radar point. The points has 2.5 mm between each other so measured_point=80 corresponds to 0.2 meters. The maximum distance will [...]

Thank you.

yalatoom, 2024-08-12:
solved

closed 2024-08-12 by yalatoom