acconeer/acconeer-python-exploration #156: XM126 Could not start - calibration failed

ParisAtDusk, 2024-11-15:
Trying to connect to A121 via XM126 fails with the following message:

Traceback (most recent call last):
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/a121/algo/_plugins/_a121.py", line 194, in start_session
    self._start_session(self._recorder)
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/a121/algo/_plugins/processor/backend_plugin.py", line 185, in _start_session
    metadata = self.client.setup_session(session_config)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 215, in setup_session
    setup_response = self._server_stream.wait_for_message(a121_messages.SetupResponse)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 68, in wait_for_message
    self._message_handler(message)
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 245, in _handle_messages
    raise ServerError(msg)
acconeer.exptool._core.communication.client.ServerError: ErroneousMessage(message='RSS, calibration failed')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/app/new/backend/_backend.py", line 230, in process_generator
    model.execute_task(task)
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/app/new/backend/_model.py", line 68, in execute_task
    plugin_task(**kwargs)
  File "/home/kita/.local/lib/python3.12/site-packages/acconeer/exptool/a121/algo/_plugins/_a121.py", line 201, in start_session
    raise HandledException(msg) from exc
acconeer.exptool.app.new._exceptions.HandledException: Could not start

None of the provided examples work. The "example_service" app outputs the following on uart:

00:00:00.169 (I) (sensor_status_a121) ------ BEGIN ------ Sensor Status Log ------ BEGIN ------
00:00:00.178 (I) (sensor_status_a121) [1] Asic identity:: 0x1210
00:00:00.183 (I) (sensor_status_a121) [1] Asic error: 0x0000
00:00:00.189 (I) (sensor_status_a121) [1] Sensor stopped@13
00:00:00.194 (I) (sensor_status_a121) [1] Scratchpad config 0: 0x0000
00:00:00.200 (I) (sensor_status_a121) [1] Scratchpad config 1: 0x0000
00:00:00.206 (I) (sensor_status_a121) [1] Scratchpad config 2: 0x0000
00:00:00.213 (I) (sensor_status_a121) [1] Scratchpad config 3: 0x0000
00:00:00.219 (I) (sensor_status_a121) [1] Scratchpad config 4: 0x0000
00:00:00.225 (I) (sensor_status_a121) [1] Scratchpad config 5: 0x0000
00:00:00.231 (I) (sensor_status_a121) [1] Scratchpad config 6: 0x0000
00:00:00.237 (I) (sensor_status_a121) [1] Scratchpad config 7: 0x0011
00:00:00.243 (I) (sensor_status_a121) [1] wakeup_status0: 0x0001
00:00:00.249 (I) (sensor_status_a121) [1] inputmux_wakeup_input0_config: 0x0001
00:00:00.256 (I) (sensor_status_a121) [1] inputmux_wakeup_input1_config: 0x0008
00:00:00.263 (I) (sensor_status_a121) [1] inputmux_wakeup_input2_config: 0x0000
00:00:00.270 (I) (sensor_status_a121) [1] inputmux_wakeup_input3_config: 0x0000
00:00:00.277 (I) (sensor_status_a121) [1] inputmux_wakeup_input4_config: 0x0000
00:00:00.284 (I) (sensor_status_a121) [1] inputmux_wakeup_input5_config: 0x0000
00:00:00.292 (I) (sensor_status_a121) [1] inputmux_wakeup_input6_config: 0x0000
00:00:00.299 (I) (sensor_status_a121) [1] inputmux_wakeup_input7_config: 0x0000
00:00:00.306 (I) (sensor_status_a121) [1] wakeup_input0_wakeup_config: Addr=98: 0x00e4
00:00:00.313 (I) (sensor_status_a121) [1] wakeup_input0_ack_config:    Addr=99: 0x0000
00:00:00.321 (I) (sensor_status_a121) [1] wakeup_input1_wakeup_config: Addr=100: 0x0014
00:00:00.329 (I) (sensor_status_a121) [1] wakeup_input1_ack_config:    Addr=101: 0x0000
00:00:00.336 (I) (sensor_status_a121) [1] wakeup_input2_wakeup_config: Addr=102: 0x0000
00:00:00.344 (I) (sensor_status_a121) [1] wakeup_input2_ack_config:    Addr=103: 0x0000
00:00:00.352 (I) (sensor_status_a121) [1] wakeup_input3_wakeup_config: Addr=104: 0x0000
00:00:00.360 (I) (sensor_status_a121) [1] wakeup_input3_ack_config:    Addr=105: 0x0000
00:00:00.367 (I) (sensor_status_a121) [1] wakeup_input4_wakeup_config: Addr=106: 0x0000
00:00:00.375 (I) (sensor_status_a121) [1] wakeup_input4_ack_config:    Addr=107: 0x0000
00:00:00.383 (I) (sensor_status_a121) [1] wakeup_input5_wakeup_config: Addr=108: 0x0000
00:00:00.391 (I) (sensor_status_a121) [1] wakeup_input5_ack_config:    Addr=109: 0x0000
00:00:00.398 (I) (sensor_status_a121) [1] wakeup_input6_wakeup_config: Addr=110: 0x0000
00:00:00.406 (I) (sensor_status_a121) [1] wakeup_input6_ack_config:    Addr=111: 0x0000
00:00:00.414 (I) (sensor_status_a121) [1] wakeup_input7_wakeup_config: Addr=112: 0x0000
00:00:00.422 (I) (sensor_status_a121) [1] wakeup_input7_ack_config:    Addr=113: 0x0000
00:00:00.429 (I) (sensor_status_a121) [1] Stack_level_status_register: 0x0003
00:00:00.436 (I) (sensor_status_a121) [1] Stack level 0: Addr=280: 356
00:00:00.443 (I) (sensor_status_a121) [1] Stack level 1: Addr=281: 1323
00:00:00.449 (I) (sensor_status_a121) [1] Stack level 2: Addr=282: 99
00:00:00.455 (I) (sensor_status_a121) [1] Stack level 3: Addr=283: 13
00:00:00.461 (I) (sensor_status_a121) [1] Stack level 4: Addr=284: 32
00:00:00.467 (I) (sensor_status_a121) [1] Stack level 5: Addr=285: 29
00:00:00.474 (I) (sensor_status_a121) [1] Stack level 6: Addr=286: 0
00:00:00.480 (I) (sensor_status_a121) [1] Stack level 7: Addr=287: 0
00:00:00.486 (I) (sensor_status_a121) [1] Stack level 8: Addr=288: 0
00:00:00.492 (I) (sensor_status_a121) [1] Stack level 9: Addr=289: 0
00:00:00.498 (I) (sensor_status_a121) [1] Stack level 10: Addr=290: 0
00:00:00.504 (I) (sensor_status_a121) [1] Stack level 11: Addr=291: 0
00:00:00.510 (I) (sensor_status_a121) [1] Stack level 12: Addr=292: 0
00:00:00.517 (I) (sensor_status_a121) [1] Stack level 13: Addr=293: 0
00:00:00.523 (I) (sensor_status_a121) [1] Stack level 14: Addr=294: 0
00:00:00.529 (I) (sensor_status_a121) [1] Stack level 15: Addr=295: 0
00:00:00.535 (I) (sensor_status_a121) [1] Interrupt status register: 0x0018
00:00:00.542 (I) (sensor_status_a121) ------- END ------- Sensor Status Log ------- END -------

A121server firmware version: 1.8.1
Acconeer exploration Tool version: latest from github

AndersBuhl, 2024-11-19:
Hi,

Seems to be an issue with the sensor. Could you provide the full log when running example_bring_up?

BR,
Anders

ParisAtDusk, 2024-11-19:
Hi, 

After a few more hours of fiddling we found that the onboard step-down converter has shorted Vin and SW pins sending 5V to the A121 radar chip. Somehow after connecting it to benchtop PSU and setting 1.8V it's still alive, but we do not know if the output values are correct. Sometimes it seems to work fine, but it struggles at close distance (below ~0.5m) and shows random distances to objects. Is there a way to verify that the sensor is fully functional? I can provide the log of example_bring_up in a few days. If there's anything else you need, let me know. 

Thank you in advance, 
Krzysztof

jbjornstedt, 2024-12-10:
This is not a problem with the Acconeer Exploration Tool application.
Please file a ticket on our Customer Helpdesk if you need more assistance: https://support.acconeer.com/support/solutions

Best Regards,
Johan

closed 2024-12-11 by jbjornstedt