acconeer/acconeer-python-exploration #157: A121 Client open timeout

misisnik, 2024-12-05:
**Describe the bug**
When freshly powered device, no problem with client open:
`client = a121.Client.open(serial_port=args.serial_port, flow_control=False)`
then normal operation of SparseIQ, starting the session, getting the data.....
after finish - program close `client.close()`

Fresh Python script start and no power reset of the device. Client.open raise error:
`acconeer.exptool._core.communication.links.buffered_link.LinkError: recv timeout`

Possile to fix this by multiple connection (second time is possible to connect without problem) 
`    er = None
    for i in range(2):
        try:
            client = a121.Client.open(serial_port=args.serial_port, flow_control=False)
            break
        except Exception as e:
            er = e
            time.sleep(0.01)
    else:
        raise er`

**Installed packages**
```
absl-py==2.0.0
acconeer-exptool==7.13.2
addict==2.4.0
astunparse==1.6.3
attributes-doc==0.3.0
attrs==23.1.0
cachetools==5.3.2
cbor2==5.6.5
certifi==2023.7.22
charset-normalizer==3.3.2
colorama==0.4.6
commonmark==0.9.1
contourpy==1.1.1
cycler==0.12.1
exceptiongroup==1.1.1
flatbuffers==23.5.26
fonttools==4.43.1
gast==0.4.0
google-auth==2.23.4
google-auth-oauthlib==1.0.0
google-pasta==0.2.0
grpcio==1.59.2
h5py==3.10.0
idna==3.4
importlib-metadata==6.8.0
importlib-resources==6.1.0
keras==2.13.1
keyboard==0.13.5
kiwisolver==1.4.5
libclang==16.0.6
Markdown==3.5.1
MarkupSafe==2.1.3
matplotlib==3.7.3
numpy==1.24.3
oauthlib==3.2.2
opt-einsum==3.3.0
packaging==23.2
pandas==2.0.3
Pillow==10.1.0
platformdirs==4.3.6
playsound==1.2.2
protobuf==4.25.0
pyasn1==0.5.0
pyasn1-modules==0.3.0
pygments==2.18.0
pyparsing==3.1.1
PyQt5==5.15.10
PyQt5-Qt5==5.15.2
PyQt5-sip==12.13.0
pyqtgraph==0.13.3
pyserial==3.5
PySide6==6.6.0
PySide6-Addons==6.6.0
PySide6-Essentials==6.6.0
python-dateutil==2.8.2
pytz==2024.2
pyusb==1.2.1
PyYAML==6.0.1
requests==2.31.0
requests-oauthlib==1.3.1
result==0.16.1
rich==12.6.0
rsa==4.9
scipy==1.10.1
shiboken6==6.6.0
six==1.16.0
tensorboard==2.13.0
tensorboard-data-server==0.7.2
tensorflow==2.13.0
tensorflow-estimator==2.13.0
tensorflow-intel==2.13.0
tensorflow-io-gcs-filesystem==0.31.0
termcolor==2.3.0
textual==0.1.18
typing-extensions==4.5.0
tzdata==2024.2
urllib3==2.0.7
werkzeug==3.0.1
wrapt==1.15.0
XlsxWriter==3.2.0
zipp==3.17.0
```

**Environment (please complete the following information):**
 - Windows 10
 - Python 3.9
 - Custom HW (M4 MCU) with Acconeer module server SDK 1.8.0
 - Acconeer exptool version: 7.13.2

**Additional context**
Haven't seen this behavior before update
Tested wiht SDK 1.2.0, 1.5.0 and 1.8.0

anupcherukuri, 2024-12-07:
I am seeing a similar issue. ET 7.13.2
**Binary flashed** : acconeer_xc120_exploration_server_a121-v1_8_1

**Error I am seeing** - Recv Timeout

**Detailed Error**

Traceback (most recent call last):
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_model.py", line 96, in connect_client
    self.client = client_factory()
                  ^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/client.py", line 51, in open
    return super().open(
           ^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/client.py", line 88, in open
    return subclass.open(  # type: ignore[no-any-return, attr-defined]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 93, in open
    return cls(client_info=client_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 117, in __init__
    self._server_info = self._retrieve_server_info()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 142, in _retrieve_server_info
    system_info_response = self._server_stream.wait_for_message(messages.SystemInfoResponse)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 64, in wait_for_message
    for message in self._stream:
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 83, in _get_stream
    self._error_callback(e)
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 138, in _close_before_reraise
    raise exception
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 81, in _get_stream
    header_in_bytes = self._link.recv_until(self.protocol.end_sequence)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/links/serial_link.py", line 159, in recv_until
    raise LinkError(msg)
acconeer.exptool._core.communication.links.buffered_link.LinkError: recv timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_backend.py", line 230, in process_generator
    model.execute_task(task)
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_model.py", line 60, in execute_task
    builtin_task(**kwargs)
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_model.py", line 102, in connect_client
    raise HandledException(msg)
acconeer.exptool.app.new._exceptions.HandledException: Failed to connect:
recv timeout

jbjornstedt, 2024-12-10:
It sounds like the firmware in your module is out of sync as it waits for a line break before it starts processing the commands sent from the client.
On our reference module, XM125, we use flow control of the UART to make sure we don't miss any data. Would it be possible for you to use that on your custom HW to make sure that is not the problem?

Best Regards,
Johan

anupcherukuri, 2024-12-10:
@jbjornstedt I am using the firmware flashed using the Exploration tool. I get the latest binary by signing in into the account.
The hardware I am using is the XM125 development kit, did not modify anything.
It used to work fine before, recently started seeing this issue.
Can you please recommend what I can try, if I need to try out another firmware or exploration tool version, please let me know.

Thank you

jbjornstedt, 2024-12-10:
@anupcherukuri You don't have the same issue as the OP.
You have flashed the XC120 binary onto your XM125 which will not work.
Make sure you select XM125 in the 'Flash' dialog before downloading.

Best Regards,
Johan

anupcherukuri, 2024-12-11:
I flashed the XM125 Binary but still having the same issue
[image]
Traceback (most recent call last):
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_model.py", line 96, in connect_client
    self.client = client_factory()
                  ^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/client.py", line 51, in open
    return super().open(
           ^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/client.py", line 88, in open
    return subclass.open(  # type: ignore[no-any-return, attr-defined]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 93, in open
    return cls(client_info=client_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 117, in __init__
    self._server_info = self._retrieve_server_info()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 142, in _retrieve_server_info
    system_info_response = self._server_stream.wait_for_message(messages.SystemInfoResponse)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 64, in wait_for_message
    for message in self._stream:
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 83, in _get_stream
    self._error_callback(e)
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/a121/_core/communication/exploration_client.py", line 138, in _close_before_reraise
    raise exception
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/message_stream.py", line 81, in _get_stream
    header_in_bytes = self._link.recv_until(self.protocol.end_sequence)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/_core/communication/links/serial_link.py", line 159, in recv_until
    raise LinkError(msg)
acconeer.exptool._core.communication.links.buffered_link.LinkError: recv timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_backend.py", line 230, in process_generator
    model.execute_task(task)
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_model.py", line 60, in execute_task
    builtin_task(**kwargs)
  File "/home/anup/anaconda3/lib/python3.11/site-packages/acconeer/exptool/app/new/backend/_model.py", line 102, in connect_client
    raise HandledException(msg)
acconeer.exptool.app.new._exceptions.HandledException: Failed to connect:
recv timeout

Can you please check and help me out with the issue?

jbjornstedt, 2024-12-11:
@misisnik 
Good to hear that you've found the root cause.

jbjornstedt, 2024-12-11:
@anupcherukuri 
Some other users have had problems with the USB connector of their XE125 and when they flipped the USB-C cable it worked. Could you try that and see if it works for you?

If not I urge you to file a ticket on our Customer Helpdesk if you need more assistance since this is not a problem with the Acconeer Exploration Tool application: https://support.acconeer.com/support/solutions

closed 2025-01-15 by jbjornstedt