acconeer/acconeer-python-exploration #163: example program bug

nofallinto, 2025-04-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Traceback**
If applicable, add any exception tracebacks.

**Installed packages**
If applicable, add a list of installed packages. Obtain the list with `pip freeze`.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
**Environment (please complete the following information):**
 - Host OS: [window10]
 - Acconeer hardware: [STM32L431RCT6]
 - Acconeer software version: [e.g. 2.0.0]
 - Exploration version: [e.g. e78bdc6 or v3.0.2]

**Additional context**
Add any other context about the problem here.

sub     sp, #76 @ 0x4c
this only 76 byte but next:

movs    r2, #236        @ 0xec
08005bfc:   movs    r1, #0
08005bfe:   ldr     r0, [sp, #112]  @ 0x70
08005c00:   bl      0x800ba4c <memset>
your memset use len = 236 so  my main stack is broken. my stack Top pointer is 0x2000c000

Running example program : In the file example_service.c and func: acc_example_service()

The call order is: acc_example_service -> do_sensor_calibration_and_prepare -> acc_sensor_calibrate

jbjornstedt, 2025-04-01:
This is not a problem with the Exploration Tool.
Please create a ticket here instead: https://support.acconeer.com/support/home

nofallinto, 2025-04-02:
> This is not a problem with the Exploration Tool. Please create a ticket here instead: https://support.acconeer.com/support/home

OK thank you
Do I need to purchase your board to use the discovery tool? I only have A121 RADAR SENSOR

jbjornstedt, 2025-04-10:
We recommend using Exploration Tool with one of our EVKs

closed 2025-04-10 by jbjornstedt