acconeer/acconeer-python-exploration #164: Data cannot be viewed

nofallinto, 2025-04-09:
Why can I use your exploration tool to connect, but I can't see the data? The sdk I use is xm125.
The program I'm running is

[image]

nofallinto, 2025-04-09:
[image]

nofallinto, 2025-04-09:
When I debug, I find that the mcu does not receive the host message at all, and the data buffer is still 0.
[image]

This is my serial port configuration, and I am not using hardware flow control, I guess this should not be the cause of the problem?
[image]

rastikw, 2025-04-09:
Downgrading to exptool to 7.14.3 and module fw to 1.19 works. But 7.15.0 doesn't.

nofallinto, 2025-04-10:
[image]
I wonder if it is necessary to download this firmware? The program I downloaded using cubeide should also contain this firmware?

closed 2025-04-10 by jbjornstedt