acconeer/acconeer-python-exploration #166: acconeer.exptool.app QGraphicsItem::itemTransform: null pointer passed

knightsmiles, 2025-06-05:
I get the same failure on both my Windows 10 and Ubuntu machines when trying to run python -m acconeer.exptool.app 
The GUI loads, however; there is no graphical presentation of the results from the XE125. Then when I look at the Linuc/Windows terminal, it just shows:
(venv) scott@scott-Inspiron-5400-AIO:~/projects/Accioneer$ python -m acconeer.exptool.app
QGraphicsItem::itemTransform: null pointer passed
QGraphicsItem::itemTransform: null pointer passed
QGraphicsItem::itemTransform: null pointer passed
QGraphicsItem::itemTransform: null pointer passed
QGraphicsItem::itemTransform: null pointer passed
QGraphicsItem::itemTransform: null pointer passed

Zycenios, 2025-06-07:
I had this exact same issue, just got my XE121 and XC120 a few days ago and was getting that error, both running on my local Python 3.11.1 installation and in the portable 3.9.10 environment included with the SDK.

Fortunately I tried downgrading PySide6 to version 6.8.3 and it started working (version was previously 6.9.1). Give this a shot:
```
python -m pip uninstall PySide6  
python -m pip install PySide6==6.8.3
```

I tried other things like other versions of the exptool and purging the config like recommended in the docs, and this is all that worked for me.

knightsmiles, 2025-06-07:
Thank you for the info. I'm going to try the pyside downgrade.

MruthyunjayaRM, 2025-06-16:
Thank you. Bring PySide6 to 6.8.3 worked for me

jbjornstedt, 2025-06-25:
Thank you for reporting this.
This should be fixed in the newly released v7.16.1