acconeer/acconeer-python-exploration #173: A111 Sparkfun on Raspberry Pi 5 — `libft4222.LibFT4222Error: DEVICE_NOT_FOUND` even when using SPI backend

MindyZHAOMinzhu, 2025-08-11:
**Description:**
When running the A111 examples with `-spi` on Raspberry Pi 5 (Bookworm), the exploration tool attempts to use the **FT4222** backend instead of the `spidev` backend, resulting in:

```
acconeer.exptool.libft4222.LibFT4222Error: Error 2: DEVICE_NOT_FOUND
```

This occurs even when:

* `A111_SPI_BACKEND=spidev` is exported
* `A111_SPI_DEVICE=/dev/spidev0.0` is set
* `A111_EN_PIN` and `A111_INT_PIN` are set
* `/dev/spidev0.0` is present and functional

**What I have tried:**

1. Raspberry Pi 5, Raspberry Pi OS Bookworm 64-bit, SPI enabled in `/boot/firmware/config.txt`:

   ```
   dtparam=spi=on
   dtoverlay=spi0-0cs
   ```
2. Install `acconeer-exptool==2.15.4` in a virtualenv.
3. Connect SparkFun A111 Breakout to Pi 5 GPIO header (default pins).
4. Run:

   ```bash
   export A111_SPI_BACKEND=spidev
   export A111_SPI_DEVICE=/dev/spidev0.0
   export A111_EN_PIN=27
   export A111_INT_PIN=25
   python examples/a111/services/envelope.py -spi -vv
   ```
5. Observe that `libft4222` is still imported and used, causing the error above.

**Expected behavior:**
When `-spi` is used (and/or `A111_SPI_BACKEND=spidev` is set), the tool should use the `spidev` backend and not attempt to load the FT4222 driver.

**System details:**

* **Board:** Raspberry Pi 5, 8 GB
* **OS:** Raspberry Pi OS Bookworm (64-bit)
* **acconeer-exptool:** 2.15.4
* **Python:** 3.11.2
* **Hardware:** SparkFun A111 Pulsed Radar Breakout
* `/dev/spidev0.0` is present and can be opened via `spidev` Python module.

vackraetraed, 2025-08-12:
Hi @MindyZHAOMinzhu 

In order to use the A111 Sparkfun with our Exploration Tool, you would have to download the ACCONEER RPI SPARKFUN A111 SDK from our developer site: https://developer.acconeer.com/home/a111-docs-software/sparkfun/
Then run the "acc_exploration_server_a111" executable on your Raspberry.
To run it on the 64 bit arch, there are instructions in the readme in the doc folder.

Once the Exploration Server is up and running, you can run the python scripts in a separate terminal.
However, the SPI is used by the Exploration Server binary to talk with the radar sensor.
So you have to connect using socket and "localhost" as hostname to connect to the Exploration Server application running in the other terminal session.

vackraetraed, 2025-08-12:
Hello @MindyZHAOMinzhu 

For questions 1 and 2, there is no know versions of either that should be matched.

However, I can see that earlier in your Exploration Server log, you get the error "Config, invalid sensor_id" when it received a "Setup" command instead.
What command did you run then?

Would be interesting with a full log from the Exploration Server.

1. Instead of running the Exploration Server, you can run "example_bring_up" and "example_service_envelope", just to verify that the software can communicate with the radar and get radar data from the sensor
2. Make sure the Sparkfun board is connected to the Raspberry when you start the Exploration Server.
3. Try running the command without --sensor 1: `python ~/Desktop/acconeer-python-exploration/examples/a111/services/envelope.py --socket 127.0.0.1 --protocol exploration` There is no logic to communicate with a specific sensor on the Sparkfun board, it always communicates with the only one that is there.

Best regards,
Anton

MindyZHAOMinzhu, 2025-08-22:
Hi Anton,

Thanks for all the support with troubleshooting! I have contacted helpdesk instead: https://acconeer.freshdesk.com/ and it is issue with Sparkfun A111. It has been solved, and many thanks!

Best,
Mindy

closed 2025-08-22 by MindyZHAOMinzhu